amaranth-lang/amaranth #822: Driver domain conflicts not detected across hierarchy

mwkmwkmwk, 2023-06-09:
The following code gets processed into nonsensical Verilog, with the only warning being `DriverConflict: Signal '(sig a)' is driven from multiple fragments: top, top.m2; hierarchy will be flattened`, instead of being rejected due to assigning a signal from two different domains:

```
from amaranth import *
from amaranth.back.verilog import convert

a = Signal()
m1 = Module()
m1.d.sync += a.eq(0)
m2 = Module()
m2.d.sync2 += a.eq(1)
m1.submodules.m2 = m2

print(convert(m1, ports=[a]))
```

whitequark, 2023-06-09:
Known issue. Fix is not planned until the entire middle-end is converted to use a netlist-style representation but the bug will be kept open in the meantime.

whitequark, 2024-02-27:
I believe this is fixed by the NIR.

closed 2024-02-27 by whitequark