amaranth-lang/amaranth #823: Fragment flattening can cause local domains to get merged

mwkmwkmwk, 2023-06-09:
```
from amaranth import *                                                                                                                                                                                                
from amaranth.back.verilog import convert                                                                                                                                                                                                

a = Signal()
d = Signal()
q1 = Signal()
q2 = Signal()
clk1 = Signal()
clk2 = Signal()
                                                                                                                                                                                                 
m1 = Module()
m1.domains.local = ClockDomain(local=True)
m1.d.sync += a.eq(0)                                
m1.d.local += q1.eq(d)
m1.d.comb += ClockSignal('local').eq(clk1)
                                                    
m2 = Module()                                       
m2.domains.local = ClockDomain(local=True)
m2.d.sync += a.eq(0)                          
m2.d.local += q2.eq(d)
m2.d.comb += ClockSignal('local').eq(clk2)
                    
mt = Module()       
mt.submodules.m1 = m1
mt.submodules.m2 = m2
                                                                                                                                                                                                 
print(convert(mt, ports=[a, d, q1, q2, clk1, clk2]))                                                                                                                                                                         
```

All fragments here will be flattened due to `a` assignments from multiple fragments, causing the `local` domains in both sub-fragments to get merged with each other.  Removing any of the assignments to `a` results in correct output.

whitequark, 2023-06-09:
Expected issue. Fix is not planned until the entire middle-end is converted to use a netlist-style representation but the bug will be kept open in the meantime.

whitequark, 2024-02-27:
@wanda-phi Is this still something we need to track?

PR #1197 (2024-03-11): hdl._ir: Remove all support for fragment flattening.
Fixes #823
closed 2024-03-11 by whitequark