amaranth-lang/amaranth #839: RecursionError in back.rtlil when compiling assignments with many operands

jfng, 2023-07-18:
Repro:
```python3
from amaranth import *
from amaranth.back import rtlil

N = 200

m = Module()
a = Signal()

a_fanin = 0
for i in range(N):
    a_fanin |= Const(1)

m.d.comb += a.eq(a_fanin)

print(rtlil.convert(m, ports=(a,)))
```

Output ([trace.txt](https://github.com/amaranth-lang/amaranth/files/12076895/trace.txt)):
```
RecursionError: maximum recursion depth exceeded
```

This example works when lowering `N` or the CPython recursion limit (1000 on my machine).

whitequark, 2023-07-18:
This is a known issue--I don't think we'll be able to address it without a rewrite of the RTLIL backend.

whitequark, 2024-02-13:
This is no longer a problem with the new IR.

closed 2024-02-13 by whitequark