amaranth-lang/amaranth #857: How to adding a new component ?

jalcim, 2023-08-06:
On the documentation and on the code i didn't find the way for adding a new component.

only find this information #781

other question, where can i find the list of component ?

whitequark, 2023-08-06:
It is unclear what are you asking. Can you rephrase your question with more detai?

jalcim, 2023-08-06:
on this exemple we use a Dlatch.
i want to know where this component is defined, and how add another one.

`def elaborate(self, platform):
        m = Module()

        m.d.comb += self.ovf.eq(self.count == self.limit)`

whitequark, 2023-08-06:
What kind of component are you aiming to add?

jalcim, 2023-08-06:
All and nothing i work with a friend who start to use amaranth, and i provide him verilog modules, we want know how he can use it.

For example i want send him a counter, or multiplexer, or usb controler xD

jalcim, 2023-08-06:
all what i find is that.
https://github.com/amaranth-lang/amaranth/blob/b9520050d115cc1c4321d1777cd16c1c05413e19/amaranth/hdl/dsl.py#L70

they have no documentation about standard-library.

whitequark, 2023-08-06:
The documentation for language constructs is [here](https://amaranth-lang.org/docs/amaranth/latest/lang.html), though it's not complete at the moment.

jalcim, 2023-08-06:
> The documentation for language constructs is [here](https://amaranth-lang.org/docs/amaranth/latest/lang.html), though it's not complete at the moment.

I know i have read it.
I wanna know how to add a new one.
I have read all the documentation, and search in the source code.

whitequark, 2023-08-06:
Adding new language constructs (under `amaranth.hdl.*`) is an advanced topic that we aren't going to cover in the documentation even when it's complete. Unless you give more details about what you're trying to achieve I'm afraid you're on your own.

jalcim, 2023-08-06:
Its not a problem for us to contribute !
I didn't ask you to do it for us.
we want understand how to do it.

Maybe after can we adding it to the documentation.
But... i feel you just didn't want to respond, and i didn't understand why.

We want add some module inside amaranth, and we search how to do it, because its not a simple software architecture.

Maybe can you provide me a starting point ?
For exemple m.d.comb.

Because without that is to limited, i didn't understand why this part is hide...

whitequark, 2023-08-06:
If you'd like to extend the language you'll need to go through the [RFC process](https://amaranth-lang.org/rfcs/). In general, I prefer to avoid having incompatible language forks around, so making it easier to make an incompatible language fork isn't something that I want to make easier.

jalcim, 2023-08-06:
Then you refuse to help me, because you'r scary i fork your project ?

adamgreig, 2023-08-06:
@jalcim, are you asking how to use an existing verilog module in an Amaranth design, or perhaps how to write a module in Amaranth? I'm not sure if you're asking how to add a whole new concept to the language, or just how to use the language for a normal project without having to modify the language itself.

jalcim, 2023-08-06:
Its a good question !
At this step i didn't know and explore all the way !
The must simple is to use a verilog module inside amaranth, because i see its hard to write a module in amaranth.

adamgreig, 2023-08-06:
Maybe some examples would help. Here's an example of writing a simple counter in Amaranth as a module and using it, and also including another counter written in Verilog, including it, instantiating it as a module, and using it.

```python
import amaranth as am
from amaranth_boards.icebreaker import ICEBreakerPlatform

class Counter(am.Elaboratable):
    def __init__(self, n):
        # Store an n-bit register for the count value
        self.count = am.Signal(n)

        # Output a single-bit output signal
        self.overflow = am.Signal()

    def elaborate(self, platform):
        m = am.Module()
        # Count up each clock cycle
        m.d.sync += self.count.eq(self.count + 1)
        # Output overflow on cycles where the count is 0
        m.d.comb += self.overflow.eq(self.count == 0)
        return m

class Top(am.Elaboratable):
    def elaborate(self, platform):
        m = am.Module()

        # Instantiate the Amaranth counter
        m.submodules.counter = counter = Counter(16)
        # Connect its output to an LED
        m.d.comb += platform.request("led_g", 0).o.eq(counter.overflow)

        # Add some Verilog
        v = """
        module verilog_counter ( clk, cnt );
            parameter WIDTH = 8;
            input clk;
            output reg [WIDTH-1:0] cnt;

            always @(posedge clk) begin
                cnt <= cnt + 1;
            end
        endmodule
        """
        platform.add_file("counter.v", v)
        count = am.Signal(8)
        m.submodules.verilog_counter = am.Instance(
            "verilog_counter",
            p_WIDTH=count.width,
            i_clk=am.ClockSignal(),
            o_cnt=count,
        )

        # Connect the top bit of its count value to another LED.
        m.d.comb += platform.request("led_r", 0).o.eq(count[-1])

        return m

if __name__ == "__main__":
    top = Top()
    plat = ICEBreakerPlatform()
    plat.build(top)
```

jalcim, 2023-08-06:
Ok then.

1. Put verilog module in a string

jalcim, 2023-08-06:
ok then

1. Put verilog module inside a string
2. ask the platform for adding a file, since this string (its make a real file we can find ?)
3. instantiate a submodule with am.Instance and set parameter, input, output
4. you use directly the output

(I use skywater, then no platform)

amaranth file
```python
import os

import amaranth as am
from amaranth import Elaboratable, Signal, Module
from amaranth.back import verilog, rtlil
from amaranth.hdl import ir

#from amaranth_boards.icebreaker import ICEBreakerPlatform

class Top(am.Elaboratable):
    def __init__(self):
        # Store an n-bit register for the count value
        self.clk = am.Signal(1)

        # Output a single-bit output signal
        self.cnt = am.Signal(8)

    def elaborate(self, platform):
        m = am.Module()

        # Add some Verilog
        v = """
        module verilog_counter ( clk, cnt );
            parameter WIDTH = 8;
            input clk;
            output reg [WIDTH-1:0] cnt;

            always @(posedge clk) begin
                cnt <= cnt + 1;
            end
        endmodule
        """
#        platform.add_file("counter.v", v)
        count = am.Signal(8)
        m.submodules.verilog_counter = am.Instance(
            "verilog_counter",
            p_WIDTH=count.width,
            i_clk=am.ClockSignal(),
            o_cnt=count,
        )
        return m

def get_rtlil_code(machin : Elaboratable):
    name = "top"
    platform = None
    ports = [machin.clk, machin.cnt]
    emit_src = True
    strip_internal_attrs = False
    fragment = ir.Fragment.get(machin, platform).prepare(ports=ports)
    rtlil_text, name_map = rtlil.convert_fragment(
        fragment,
        name,
        emit_src=emit_src,
    )
    ENV_USERNAME = os.environ.get("USER")
    rtlil_source_text = rtlil_text.replace(ENV_USERNAME, "user")
    return rtlil_source_text
    
if __name__ == "__main__":
    machin = Top()

    rtlil_source_text = get_rtlil_code(machin)
    outfile_rtlil = "output_up_counter.rtlil"
    with open(outfile_rtlil, "w") as fd:
        fd.write(rtlil_source_text)
        print(f"'{outfile_rtlil}' filename written.")

#    plat = ICEBreakerPlatform()
#    plat.build(top)
```

verilog file
```verilog
module verilog_counter ( clk, cnt );
   parameter WIDTH = 8;
   input clk;
   output reg [WIDTH-1:0] cnt;

   always @(posedge clk) begin
      cnt <= cnt + 1;
   end
endmodule
```

inside yosys cmd

```spice
read_rtlil output_up_counter.rtlil
read_verilog count.v
procs
hierarchy
synth
write_spice out.sp
```

I think i make a mistake, because the output in spice is empty

jalcim, 2023-08-06:
Thanks you very much for the help !

closed 2023-08-06 by jalcim