amaranth-lang/amaranth #864: docs: Document Case, FSM

mcclure, 2023-08-10:
The language guide https://amaranth-lang.org/docs/amaranth/latest/lang.html#control-structures

Contains this sentence: "Amaranth provides control structures tailored for this task: If, Switch, and FSM."

It then proceeds to explain If but *not* Switch or FSM.

Wanda and crzwdjk in the Matrix explain:

> There is indeed a switch statement, you use it like "with m.Switch(discriminant):"
> ```
> with m.Switch(abc):
>   with m.Case(0x12):
>      m.d.sync += [...]
>   with m.Default():
>      [...]
> ```
> And then inside that "with m.Case(...):" where the case value can be a constant or a bit string
> you can also get fancy and match bit patterns like `m.Case("----1101----"):` or something

The guide warns "This guide is a work in progress and is seriously incomplete!" but regardless documenting at least the flow control constructs seems important.

whitequark, 2023-08-10:
> The guide warns "This guide is a work in progress and is seriously incomplete!" but regardless documenting at least the flow control constructs seems important.

As far as I recall, I was writing it top to bottom until a point where an abusive work environment became untenable (again). I have not touched it since.

whitequark, 2024-01-09:
This has been documented.

closed 2024-01-09 by whitequark