amaranth-lang/amaranth #873: Implement RFC 18: Reorganize vendor platforms

whitequark, 2023-08-22:


gatecat, 2023-08-22:
Some implementation questions I have:

  - how should the lazy loading work? just checking all the vendor modules to see which one the class is in, or having a class->module dict in the `__init__.py`? (not sure if there's a standard python paradigm here I just don't know and couldn't immediately find)
   - do we need backwards compatibility?
   - in amaranth-boards there's a bunch of `from amaranth.vendor.lattice_ecp5 import *`, this would be refactored to specific imports like `from amaranth.vendor import LatticeECP5Platform`?

whitequark, 2023-08-22:
> * having a class->module dict in the __init__.py?

I'm perfectly happy to have something like:

```python
def __getattr__(self, name):
    if name == "XilinxPlatform":
        from ._xilinx import XilinxPlatform as platform_cls
    elif name == "...":
        ...
    return platform_cls
```

> * do we need backwards compatibility?

Yes. Please leave modules like `vendor.xilinx` which just import the right class, and issue a deprecation warning. They will be removed after Amaranth 0.4 is released.

> * in amaranth-boards there's a bunch of `from amaranth.vendor.lattice_ecp5 import *`, this would be refactored to specific imports like `from amaranth.vendor import LatticeECP5Platform`?

Yes

PR #1002 (2023-12-13): Remove names for vendor platforms that were deprecated in RFC 18
Closes #873.
commit 3ed78d98 (2023-12-13): Implement RFC 18: Reorganize vendor platforms

Closes #873.
closed 2023-12-13 by whitequark