amaranth-lang/amaranth #882: Unfriendly "recursion depth exceeded" error message when forgetting `In` and `Out` on a `Component`'s signature

cr1901, 2023-08-27:
Consider the following Amaranth code, specifically the `rand` and `seq` lines in the `PageBuffer` `Component`:

```
from amaranth import Signal, Module
from amaranth.lib.wiring import Signature, In, Out, Component

class SeqSignature(Signature):
    def __init__(self):
        return super().__init__({
            "data": In(8),
            "addr": Out(10),
            "ready": Out(1),
            "valid": In(1)
        })

class RandSignature(Signature):
    def __init__(self):
        return super().__init__({
            "data": In(8),
            "addr": Out(14),
            "read_en": Out(1),
            "flush": Out(1),
            "valid": In(1)
        })

class PageBuffer(Component):
    rand: RandSignature().flip()
    seq: SeqSignature()

    def elaborate(self, plat):
        m = Module()

        return m

if __name__ == "__main__":
    m = PageBuffer()
```

On both Linux and Windows, the above snippet results in a recursion error along the lines of:

```
PS C:\msys64\home\William\Projects\FPGA\amaranth\efbutils\ufm_reader> python .\ufm_reader\page_buffer.py
Traceback (most recent call last):
  File "C:\msys64\home\William\Projects\FPGA\amaranth\efbutils\ufm_reader\ufm_reader\page_buffer.py", line 37, in <module>
    m = PageBuffer()
        ^^^^^^^^^^^^
  File "C:\msys64\home\William\Projects\FPGA\amaranth\amaranth\build\__editable__.amaranth-0.4.dev173+gf95fe45-py3-none-any\amaranth\lib\wiring.py", line 744, in __init__
    for name in self.signature.members:
                ^^^^^^^^^^^^^^
  File "C:\msys64\home\William\Projects\FPGA\amaranth\amaranth\build\__editable__.amaranth-0.4.dev173+gf95fe45-py3-none-any\amaranth\lib\wiring.py", line 761, in signature
    if (annot in (Value, Signal, Const) or
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\msys64\home\William\Projects\FPGA\amaranth\amaranth\build\__editable__.amaranth-0.4.dev173+gf95fe45-py3-none-any\amaranth\lib\wiring.py", line 489, in __eq__
    return other == self
           ^^^^^^^^^^^^^
  File "C:\msys64\home\William\Projects\FPGA\amaranth\amaranth\build\__editable__.amaranth-0.4.dev173+gf95fe45-py3-none-any\amaranth\lib\wiring.py", line 489, in __eq__
    return other == self
           ^^^^^^^^^^^^^
  File "C:\msys64\home\William\Projects\FPGA\amaranth\amaranth\build\__editable__.amaranth-0.4.dev173+gf95fe45-py3-none-any\amaranth\lib\wiring.py", line 489, in __eq__
    return other == self
           ^^^^^^^^^^^^^
  [Previous line repeated 495 more times]
RecursionError: maximum recursion depth exceeded in comparison
```

If I put `In()` and `Out` around the signature annotations, the error goes away:

```
class PageBuffer(Component):
    rand: In(RandSignature().flip())
    seq: Out(SeqSignature())
```

A more friendly error message along the lines of "you forgot `In` and `Out`" would be helpful to avoid debugging a misleading stack overflow error. On the other hand, RFC 2 [contains](https://amaranth-lang.org/rfcs/0002-interfaces.html) code examples where `In` and `Out` are omitted from the signature annotation. So it's not clear to me whether the initial code snippet should be erroring out or not.

whitequark, 2023-08-27:
> A more friendly error message along the lines of "you forgot `In` and `Out`" would be helpful to avoid debugging a misleading stack overflow error.

This diagnostic already exists! In fact, the code that's giving you a stack overflow is literally attempting (and failing) to print the error message you request :p`

whitequark, 2023-08-31:
Minimized:

```python
from amaranth.lib.wiring import Signature, Component

class PageBuffer(Component):
    rand: Signature({}).flip()

PageBuffer()
```

commit 44d5fac0 (2023-08-31): lib.wiring: fix equality of `FlippedSignature` with other object.

Fixes #882.
PR #885 (2023-08-31): lib.wiring: fix equality of `FlippedSignature` with other object
Fixes #882.
closed 2023-08-31 by whitequark