amaranth-lang/amaranth #899: Memory generates an empty Verilog wrport module that Vivado considers a black box

daniestevez, 2023-09-03:
I've noticed that after updating to the current https://github.com/amaranth-lang/amaranth/commit/f135226a79fddf5caf894030ac7f49995774c3a7 main branch of amaranth, and OSS CAD suite 20230903 (relevant, since the problem might actually be in Yosys), my amaranth code is failing to build on Vivado.

[This is the code in question](https://github.com/maia-sdr/maia-sdr/blob/79eac6f1538b93fbe52e1f008c40714cccf18241/maia-hdl/maia_hdl/fft.py#L168). It uses an amaranth `Memory` with `rdport` and `wrport`. These new versions of amaranth and Yosis generate some empty modules that look like this
```
(* \amaranth.hierarchy  = "top.spectrometer.fft.bfly0.bfly0.wrport" *)
(* generator = "Amaranth" *)
module wrport();
  wire \$empty_module_filler ;
endmodule
```
and which get instantiated in the module (`\bfly0$1`) that contains the `Memory`. The older versions of amaranth and Yosis that I was using before, which are from 2023-06-10, don't do this. They don't generate the empty `wrport` modules, and in fact the module `\bfly0$1` they generate doesn't instantiate any modules.

Vivado is failing to build due to these empty modules. The error it gives is
```
ERROR: [DRC INBB-3] Black Box Instances: Cell 'i_system_wrapper/system_i/maia_sdr/inst/spectrometer/fft/bfly0/bfly0/wrport' of type 'system_maia_sdr_0_wrport' has undefined contents and is considered a black box.  The contents of this cell must be defined for opt_design to complete successfully.
```
so apparently the `wire \$empty_module_filler` is not enough to make Vivado not consider these modules as black boxes.

More information (probably not so helpful) in https://github.com/maia-sdr/maia-sdr/pull/72, where I first noticed this problem.

I can try to make a minimal example that shows this problem if it helps.

daniestevez, 2023-09-03:
Here is a minimal example.

```python
#!/usr/bin/env python3

from amaranth import *
import amaranth.back.verilog

class TestMemory(Elaboratable):
    def __init__(self, width=16, awidth=8):
        self.w = width
        self.aw = awidth
        self.raddr = Signal(self.aw)
        self.waddr = Signal(self.aw)
        self.rdata = Signal(self.w)
        self.wdata = Signal(self.w)
        self.wren = Signal()

    def elaborate(self, platform):
        m = Module()
        mem = Memory(width=self.w, depth=2**self.aw,
                     attrs={'ram_style': 'block'})
        m.submodules.rdport = rdport = mem.read_port(
                transparent=False)
        m.submodules.wrport = wrport = mem.write_port()
        m.d.comb += [
            rdport.en.eq(1),
            wrport.en.eq(self.wren),
            rdport.addr.eq(self.raddr),
            wrport.addr.eq(self.waddr),
            wrport.data.eq(self.wdata),
            self.rdata.eq(rdport.data),
        ]
        return m

with open('test_memory.v', 'w') as f:
    top = TestMemory()
    f.write(amaranth.back.verilog.convert(
        top, ports=[top.raddr, top.waddr, top.rdata, top.wdata, top.wren]))
```

The old version of the toolchain generates
```verilog
/* Generated by Yosys 0.30+1 (git sha1 236e15f3b, clang 10.0.0-4ubuntu1 -fPIC -Os) */

(* \amaranth.hierarchy  = "top" *)
(* top =  1  *)
(* generator = "Amaranth" *)
module top(waddr, rdata, wdata, wren, clk, rst, raddr);
  (* src = "/usr/local/lib/python3.10/dist-packages/amaranth/hdl/ir.py:527" *)
  input clk;
  wire clk;
  (* src = "/hdl/./test_memory.py:20" *)
  wire [7:0] mem_r_addr;
  (* src = "/hdl/./test_memory.py:20" *)
  wire [15:0] mem_r_data;
  (* src = "/hdl/./test_memory.py:20" *)
  wire mem_r_en;
  (* src = "/hdl/./test_memory.py:22" *)
  wire [7:0] mem_w_addr;
  (* src = "/hdl/./test_memory.py:22" *)
  wire [15:0] mem_w_data;
  (* src = "/hdl/./test_memory.py:22" *)
  wire mem_w_en;
  (* src = "/hdl/./test_memory.py:10" *)
  input [7:0] raddr;
  wire [7:0] raddr;
  (* src = "/hdl/./test_memory.py:12" *)
  output [15:0] rdata;
  wire [15:0] rdata;
[... 341 lines cut ...]
  initial _0_ = 16'h0000;
  assign mem_r_data = _0_;
  wrport wrport (
  );
  assign rdata = mem_r_data;
  assign mem_w_data = wdata;
  assign mem_w_addr = waddr;
  assign mem_r_addr = raddr;
  assign mem_w_en = wren;
  assign mem_r_en = 1'h1;
endmodule

(* \amaranth.hierarchy  = "top.wrport" *)
(* generator = "Amaranth" *)
module wrport();
  wire \$empty_module_filler ;
endmodule
```

Looking at the diff, besides irrelevant things such as the old version initializing the memory to zeros and calling it `mem` and the new version not initializing it and calling it `\rdport`, the only difference I see is that the new version includes
```
wrport wrport (
);
```
which seems to come out of nowhere.

daniestevez, 2023-09-03:
The problem appears to be in amaranth rather than Yosys. I'm now running the current amaranth main with the 20230610 Yosys and the empty `wrport` module is generated.

daniestevez, 2023-09-03:
I've bisected the problem to 8c4a15ab92af10d5640be4d15f27f0e8843ec154.

(Sorry for writing a whole bunch of comments as I debug this further).

whitequark, 2023-09-04:
@daniestevez Thanks! Feel free to stop debugging, I know what change it was (though you figured it out) and how to address it.

whitequark, 2023-09-05:
It's actually a bug I introduced back in 2020.

whitequark, 2023-09-05:
> besides irrelevant things such as the old version initializing the memory to zeros and [...] and the new version not initializing it [...]

This isn't irrelevant at all! That's actually another bug.

commit 525c7e2b (2023-09-05): back.rtlil: do not translate empty subfragments at all.

It was thought previously (by me) that adding a wire that does
nothing to an empty subfragment is enough to prevent it from being
treated as a blackbox. This is enough for Yosys but not Vivado.
Another workaround could probably be used that satisfies both, but
instead let's just not translate any empty subfragments.

This doesn't account for the case of the empty toplevel, but that
does not seem worth addressing.

Fixes #899.
closed 2023-09-05 by whitequark