amaranth-lang/amaranth #914: Potential inconsistency in lib.wiring

nelgau, 2023-09-27:
I believe I've found an inconsistency in `lib.wiring` while creating a component that uses `amaranth-soc`'s Wishbone interface. It appears to be an issue with how Amaranth creates interfaces for flipped signatures and not with `amaranth-soc` itself. The original issue can be reproduced like so:

```python
from amaranth import *
from amaranth.lib.wiring import In, Out, Component
from amaranth_soc import wishbone

class BusComponent(Component):
    output: Out(wishbone.Signature(addr_width=32, data_width=32))
    input:  In(wishbone.Signature(addr_width=32, data_width=32))

c = BusComponent()

print(type(c.output))
print(type(c.input))

c.output.addr_width
c.input.addr_width
```

Result:

```

    c.input.addr_width
AttributeError: 'Interface' object has no attribute 'addr_width'
```

The `c.output` attribute is a `wishbone.Interface`. This is expected because `wishbone.Signature` overrides the create method [here](https://github.com/amaranth-lang/amaranth-soc/blob/c5e1e6606a9452601c274e3e48cda8dbcf353b96/amaranth_soc/wishbone/bus.py#L180-L190) to return a `wishbone.Interface`. On the other hand, the `c.input` attribute is a `wiring.Interface`.

It doesn't appear to affect interfaces flipped _after_ they're created, only 

```

nelgau, 2023-09-27:
One possible solution would be to implement `FlippedSignature.create` to return the flipped interface constructed from the unflipped signature. After implementing it as below, the above reproductions appear to work correctly.

```python
class FlippedSignature:
    # ...
    def create(self, *, path=()):
        return flipped(self.__unflipped.create(path=path))
```

Is there a better way to achieve the same effect?

whitequark, 2023-09-27:
I think this is just a bug! We will look into your proposed solution. Thank you for filing it ^^

commit a90bc7b9 (2023-09-27): lib.wiring: create flipped interface from flipped signature.

Fixes #914.
closed 2023-09-27 by whitequark