amaranth-lang/amaranth #917: `lib.wiring.connect` does not take custom `Signature.__eq__` overrides into account

whitequark, 2023-09-27:
This is an issue if e.g. two components have the same members in their interfaces, but different latency requirements.

This can likely be solved by checking for structural equality first (with the extensive diagnostics we currently have in `connect`) and then *also* running `__eq__` on the appropriately flipped interfaces.

wanda-phi, 2023-11-22:
There is a problem here: with the current design, it is entirely possible to connect two (or more) interfaces whose signatures are not just flipped versions of each other — you can connect `(Out, In, In)` together with `(In, Out, In)` and `(In, In, Out)` for example. No amount of signature flipping will make these three signatures compare equal, so I see no way to delegate to `__eq__` without breaking this functionality.

After some brief discussion with @whitequark, we think a better approach may be to instead build on the annotation mechanism introduced by amaranth-lang/rfcs#30 and introduce a subtype of annotation that needs to compare equal between signatures for them to be considered connectable. Since this can be introduced later in a backward-compatible way, this issue will be moved to a later milestone.

whitequark, 2023-11-22:
I will also close this as "not planned" for the time being, noting that we may reopen this issue at a later point if the need for this feature is readily apparent (i.e. if we get bugs that could be prevented by this feature). Initially I assessed this as an "easy, low-cost modification" which it clearly isn't because it's impossible, and we can make some other, more complex modification at a later point when we have resources to do so.

closed 2023-11-22 by whitequark