amaranth-lang/amaranth #926: Failure of sign-extension for bitwise operations

tilk, 2023-10-05:
Take the following test code:

```python
from amaranth import *
from amaranth.sim import *

class Test(Elaboratable):
    def __init__(self):
        self.sig1 = Signal(32)
        self.sig2 = Signal(32)

    def elaborate(self, platform):
        m = Module()

        c = Signal(32)
        m.d.comb += c.eq(0x12345678)

        m.d.comb += self.sig1.eq(c & ~0x101)
        m.d.comb += self.sig2.eq(c & C(~0x101, 32))

        return m

if __name__ == "__main__":
    dut = Test()

    sim = Simulator(dut)

    def process():
        yield Settle()
        print("%x" % (yield dut.sig1))
        print("%x" % (yield dut.sig2))

    sim.add_process(process)

    sim.run()
```

On Amaranth 0.3, both printed values are the same. However, on current Amaranth, I get the following output:
```
278
12345678
```
This is unexpected for me. I was under impression that Amaranth operators are designed to mimic Python semantics as much as possible in order to least surprise the programmer. Python's response is:
```
>>> print("%x" % (0x12345678 & ~0x101))
12345678
```
Which leads me to believe that the behavior in Amaranth 0.3 is the correct one, and the new one is a bug.

whitequark, 2023-10-05:
That sounds like a bug. Could you bisect please?

wanda-phi, 2023-10-05:
It bisects to 99417d6499b006a172d5b8cba413fd6181737374 , which indeed seems wrong. I think it should be calling `sign()` and not `mask()`?

tilk, 2023-10-05:
Just to be sure, I checked the result of synthesis to Verilog. In both tested versions, both outputs are 0x12345678.

commit ccf7aaf0 (2023-10-05): sim._pyrtl: fix masking for bitwise operands and muxes.

Fixes #926.
closed 2023-10-05 by whitequark