amaranth-lang/amaranth #931: Something I did made Amaranth emit invalid Verilog

mcclure, 2023-10-07:
Using Amaranth (0.4.dev198+g05cb82b) to target Cyclone V (Analogue Pocket), which means generating Verilog which gets parsed by Quartus. Got an error of invalid Verilog from Quartus.

Errors contained phrases such as

> near text: "endcase";  expecting an operand
> near text: "end";  expecting "endcase"
> near text: "always";  expecting "endcase"

Repro code:

https://github.com/mcclure/analogue-core-template-amaranth/tree/control-test COMMIT [2354a5e](https://github.com/mcclure/analogue-core-template-amaranth/commit/2354a5ee55df975bdebbd1100a7ad10f087a7eab)

Build steps (contains system local paths):

```
(cd src/fpga/amaranth_core/ && python.exe -m pdm generate) && \
(cd src/fpga && /mnt/d/intelFPGA_lite/22.1std/quartus/bin64/quartus_sh.exe --flow compile ap_core)
```

Error messages:

[quartus_errors.txt](https://github.com/amaranth-lang/amaranth/files/12839385/quartus_errors.txt)

[amaranth_core.v.txt](https://github.com/amaranth-lang/amaranth/files/12839386/amaranth_core.v.txt)

I have not yet tried clearing my pdm.lock.

mcclure, 2023-10-07:
Cleared pdm.lock, upgraded to Amaranth 0.4.dev214+gc7da6c1. No apparent change. Can re-upload errors/core.v from new code if helpful.

mcclure, 2023-10-07:
Looking at the error site:

In my code, I have this complex tree of checking conditions such as "bottom_left_diag" "top_left_diag"
Based on this I set various values in the "comb" domain
The way it has been described to me amaranth comb signals "do not exist"
That is, it ought to be no actual verilog will be emitted at the point where they are set. Instead they are inlined into some other expression later
Amaranth is emitting the tree of IFs but putting nothing inside them (possibly because nothing is inside them but combs, which do not exist)

whitequark, 2023-10-07:
> The way it has been described to me amaranth comb signals "do not exist"
> That is, it ought to be no actual verilog will be emitted at the point where they are set. Instead they are inlined into some other expression later

I... don't think that's accurate at all?

mcclure, 2023-10-08:
> I... don't think that's accurate at all?

Okay. Well in any case these case statements were empty as if their contents had been optimized away.

mcclure, 2023-10-08:
In my testing, this is fixed by https://github.com/YosysHQ/yosys/pull/3992

(With that patch, my code will successfully build. When run, the code paints a black screen. Because this is the first time I have compiled it, it is as of yet unclear to me whether that is correct.)

mcclure, 2023-10-08:
Update: Fixed some logic bugs so the screen is no longer back and can confirm, yosys#3992 fixes the issue

whitequark, 2023-10-08:
@wanda-phi Is this something that was broken in Yosys since forever, or is this relatively new? Asking because we might want to update the Yosys version constraint in response to this issue.

wanda-phi, 2023-10-08:
I'm reasonably sure that has been broken since forever.

whitequark, 2023-10-08:
Thanks. In that case once the upstream PR gets merged I'll need to bump amaranth-yosys dependency and also required Yosys version.

wanda-phi, 2023-10-08:
It has already been merged.

whitequark, 2023-10-08:
Ah right. In that case I'm waiting until the next version bump so that I can do something reasonably consistent in face of [this monstrosity](https://github.com/amaranth-lang/amaranth-yosys/blob/develop/setup.py#L22).

whitequark, 2023-11-07:
Yosys 0.35 was released, so that can become our new Yosys requirement.

PR #955 (2023-11-21): back.verilog: require Yosys >=0.35
Fixes #931.

Requires https://github.com/amaranth-lang/amaranth-yosys/actions/runs/6945417785 to complete.
commit fc06dd76 (2023-11-21): back.verilog: require Yosys >=0.35.

Fixes #931.
closed 2023-11-21 by whitequark