amaranth-lang/amaranth #934: Allow `isinstance()`/`issubclass()` to work between Interface and FlippedInterface

jfng, 2023-10-13:
This works with `Signature` and `FlippedSignature`, but not with interfaces:

```python3
from amaranth import *
from amaranth.lib.wiring import *

sig   = Signature({"foo": In(unsigned(1))})
iface = Interface(sig, path=("iface",))

assert isinstance(sig.flip(), Signature)
assert isinstance(flipped(iface), Interface) # fails
```

wanda-phi, 2023-11-16:
I do not quite understand how this is supposed to work, given that `Interface` is not supposed to be special, and interfaces can be of any custom class that doesn't inherit from interface.

What if:

```
class A:
    signature = Signature({})

a = A()

print(isinstance(flipped(a), A))
print(isinstance(flipped(a), Interface))
```

Do we want the first test to be true? This cannot be done by Python overloading rules (`__instancecheck__` would need to be on `A`), making `Interface` special.

Do we want the second test to be true? That makes `Interface` even more special.

whitequark, 2023-11-21:
Yeah, I agree with Wanda: I think we cannot feasibly implement this.

closed 2023-11-21 by whitequark