amaranth-lang/amaranth #935: `Signature.is_compliant()` should explicitly require its argument to be an Interface.

jfng, 2023-10-13:
It currently works with any Python object with a matching set of attributes.

zyp, 2023-10-13:
Remember that any Python object with a signature and a matching set of attributes *is* an interface. The RFC explicitly states that they don't have to inherit from `amaranth.lib.wiring.Interface`:

> Any object can be an interface object if it has the appropriate signature property. However, an amaranth.lib.wiring.Interface class is introduced, serving two purposes: instantiating interfaces from an anonymous signature, and serving as a convenient base class for custom interface classes.

whitequark, 2023-10-13:
@jfng IIRC what we discussed was ensuring that `connect()` calls `is_compliant()` or something? Do I misremember our discussion?

whitequark, 2023-10-25:
This wasn't what we discussed and what I think we discussed seems to be already implemented, so I'm closing this issue.

whitequark, 2023-11-26:
I figured out what this issue was supposed to be about. Right now, `Signature.is_compliant(sig, obj)` only checks that `obj` has all the attributes that `sig` requires. It does not check that `obj` has a `signature` attribute and that `sig == obj.signature`. This should be fixed.

PR #971 (2023-11-27): lib.wiring: in `is_compliant(sig, obj)`, check that `obj` is an interface object with that signature
Fixes #935.

This behavior isn't clearly specified one way or another in the RFC, and the implementation in the PR feels more appropriate to me.
commit 04f90696 (2023-11-27): lib.wiring: in `is_compliant(sig, obj)`, check that `obj` is an interface object with that signature.

Fixes #935.
closed 2023-11-27 by whitequark