amaranth-lang/amaranth #936: Invalid verilog generated for 0-bit Signals

Kristopher38, 2023-10-19:
Consider this minimal example:
```Python
from amaranth import *
from amaranth.back import verilog

class Foo(Elaboratable):
    def __init__(self):
        self.bar = Signal(0)
        self.baz = Signal(1)

    def elaborate(self, platform):
        m = Module()
        m.d.comb += self.baz.eq(self.baz > self.bar)
        return m

foo = Foo()
print(verilog.convert(foo, ports=[foo.bar, foo.baz], strip_internal_attrs=True))
```

This outputs:
```
/* Generated by Amaranth Yosys 0.10.0 (PyPI ver 0.10.0.dev46, git sha1 dca8fb54a) */

module top(baz);
  wire \$1 ;
  output baz;
  assign \$1  = baz > "";
  assign baz = \$1 ;
endmodule
```

This is invalid because `""` was not declared, and even if it was `""` is not a valid verilog identifier.
Tested on Python 3.10.7 + amaranth 392ead8d00c9a130b656b3af45c21fa410268301

Kristopher38, 2023-10-19:
This was due to using ancient version of `amaranth-yosys`.

closed 2023-10-19 by Kristopher38

whitequark, 2023-10-20:
> This was due to using ancient version of `amaranth-yosys` and issue was resolved by updating to latest version.

Could you let me know how you ended up in that position? There is a version constraint on `amaranth-yosys`; did that not work with your installation method?

Kristopher38, 2023-10-20:
We have `amaranth-yosys` [pinned to a specific version](https://github.com/kuznia-rdzeni/coreblocks/blob/master/requirements.txt) that hasn't been updated in ages and install with pip.

whitequark, 2023-10-20:
Ah, this looks to be a bug in Amaranth, as the version you have is expressly permitted. Fortunately there is already an open bug about this (https://github.com/amaranth-lang/amaranth/issues/931) so this will get fixed in time for the next release.