amaranth-lang/amaranth #946: SyncFIFOBuffered raise a TypeError

tcjie, 2023-10-25:
TypeError: Trying to add <amaranth.hdl.mem.Memory object at 0x00000203FEA34A00>, which does not implement .elaborate(), as a submodule

whitequark, 2023-10-25:
Please include a snippet of code reproducing the error.

tcjie, 2023-10-25:
from amaranth import *
from amaranth.lib.fifo  import SyncFIFOBuffered
from amaranth.back import verilog

class TestFifo(Elaboratable):
    def __init__(self) -> None:
        super().__init__()

        self.wr_en = Signal()
        self.wr_data = Signal(8)
        self.wr_rdy   = Signal()

        self.rd_en = Signal()
        self.rd_data = Signal(8)
        self.rd_rdy  = Signal()

    def elaborate(self,platform):
        m = Module()

        m.submodules.fifo = fifo_m = SyncFIFOBuffered(width=8,depth=16)

        m.d.comb += [
            fifo_m.w_en.eq(self.wr_en),
            fifo_m.w_data.eq(self.wr_data),
            self.wr_rdy.eq(fifo_m.w_rdy),

            fifo_m.r_en.eq(self.rd_en),
            self.rd_data.eq(fifo_m.r_data),
            self.rd_rdy.eq(fifo_m.r_rdy)
        ]
        return m

    def ports(self):
        return [self.wr_en,self.wr_data,self.wr_rdy,self.rd_en,self.rd_data,self.rd_rdy]

if __name__ == "__main__":
    rf = TestFifo()
    with open("test_fifo.v","w") as f:
        f.write(verilog.convert(rf,ports = rf.ports(),emit_src=False ))

whitequark, 2023-10-25:
This code runs fine and creates a file called `test_fifo.v` for me on Amaranth commit 4e4085a95b50e41288615d20211c6c5a41a01193.

tcjie, 2023-10-25:
Sorry, I use the old version Amaranth

closed 2023-10-25 by whitequark