amaranth-lang/amaranth #951: Feature:  Sum of product sop(), product of sum pos()

jrmoserbaltimore, 2023-11-10:
I wrote this terrible thing:

```python
        m.comb += [
            # output 'a'
            self.Output[0].eq(all([~b,~d]) | all([~a,c]) | all([b,c])
                | all([~a,b,d]) | all([a,~c,~d]) | all([a,~b,~c])),
            # output 'b'
            self.Output[1].eq(all([~b,~d]) | all([~a,~b]) | all([~a,~c,~d])
                | all([a,~c,d]) | all([~a,c,d])),
            # output 'c'
            self.Output[2].eq(all([~a,b]) | all([a,~b]) | all([~c,d])
                | all([~a,~c]) | all([~a,d])),
            # output 'd'
            self.Output[3].eq(all([a,~c]) | all([~a,~b,~d]) | all([~b,c,d])
                | all([b,~c,d]) | all([b,c,~d])),
            # output 'e'
            self.Output[4].eq(all([~b,~d]) | all([a,b]) | all([a,c])
                | all([c,~d])),
            # output 'f'
            self.Output[5].eq(all([~c,~d]) | all([a,~b]) | all([a,c])
                | all([b,c,~d]) | all([~a,b,~c])),
            # output 'g'
            self.Output[6].eq(all([c,~d]) | all([a,~b]) | all([a,d])
                | all([~a,~b,c]) | all([~a,b,~c])),
        ]
```

(Editor's note:  I noticed a bug—a missing `[`—in the above while writing this, because this syntax suuuuuuuucks)

I propose `sop([a,~b,c], [a,~b,d], [a,b])` equivalent to `any([all([a, ~b, c]), all([a, ~b, d]), all([a,b])])`, and `pos()` an `all(any())` shorthand.  I would also be amenable to passing a single argument of a list of lists `sop([[a,~b,c], [a,~b,d], [a,b]])`

The above code then looks like this:

```python
        m.comb += [
            # output 'a'
            self.Output[0].eq(sop([~b,~d], [~a,c], [b,c], [~a,b,d], [a,~c,~d], [a,~b,~c])),
            # output 'b'
            self.Output[1].eq(sop([~b,~d], [~a,~b], [~a,~c,~d], [a,~c,d], [~a,c,d])),
            # output 'c'
            self.Output[2].eq(sop([~a,b], [a,~b], [~c,d], [~a,~c], [~a,d])),
            # output 'd'
            self.Output[3].eq(sop([a,~c], [~a,~b,~d], [~b,c,d], [b,~c,d], [b,c,~d])),
            # output 'e'
            self.Output[4].eq(sop([~b,~d], [a,b], [a,c], [c,~d])),
            # output 'f'
            self.Output[5].eq(sop([~c,~d], [a,~b], [a,c], [b,c,~d], [~a,b,~c])),
            # output 'g'
            self.Output[6].eq(sop([c,~d], [a,~b], [a,d], [~a,~b,c], [~a,b,~c])),
        ]
```

jrmoserbaltimore, 2023-11-10:
True, but you can also write `any()` and `all()` yourself.  `mux()` is another thing you can do yourself.

Sum of products/product of sums are common operations; I also considered sending a pull request for this but it was late and it's not immediately obvious to me the correct way to iterate over an array passed to a function to carry out the `sop()` or `pos()` functions.  Maybe `any([all(x) for x in inputarray])` or is that not how amaranth works?  I *think* it should work, maybe?  Probably.

Given how common the operation is, I actually went looking through the docs and codebase before making a feature request, since it wasn't obvious I wasn't just missing something.

whitequark, 2023-11-10:
> Sum of products/product of sums are common operations

That's not really the case in my experience. You're the first person who wanted a first-class sum-of-product operation that I'm aware of.

In any case, additions to the public API have to go through [the RFC process](https://amaranth-lang.org/rfcs/).

jrmoserbaltimore, 2023-11-10:
That makes sense

> > True, but you can also write `any()` and `all()` yourself. `mux()` is another thing you can do yourself.
> 
> `any()` and `all()` are standard Python functions. `Mux()` is the Amaranth equivalent of the standard Python `x if y else z` construct. `sop()` and `pos()` do not have a Python correspondence.

That makes sense.

> 
> > Sum of products/product of sums are common operations
> 
> That's not really the case in my experience. You're the first person who wanted a first-class sum-of-product operation that I'm aware of.

I'm not sure those two things are equivalent, but fair enough.  Large functions are impossible to k-map anyway since you need to do them in an n-dimensional hypercube for 2n inputs (k-mapping is an optimization technique used to minimize logic by finding the optimal sum-of-products/product-of-sums equation; the other method is Quine-McClusky, which is np-hard because you need to solve the set covering problem, and that outputs sop/pos too).

> 
> In any case, additions to the public API have to go through [the RFC process](https://amaranth-lang.org/rfcs/).

Thanks, I'll look through that.

whitequark, 2023-11-10:
Amaranth primarily targets FPGAs and ASICs and the open toolchain uses AIG-based synthesis for either, meaning neither K-maps nor Quine-McClusky are relevant at all for any of the main flows. CPLDs used to use those heavily but there is no CPLD toolchain explicitly supported at the moment (I think you could do some weird things with ISE but I'm not aware of anyone who has done that on a more than curiosity basis).

Even if K-maps were used, Amaranth is a high-level frontend that leaves all of the mapping work to the underlying toolchain, so even then I don't see the relevance of SOP/POS in the frontend.

jfng, 2023-11-10:
> > Sum of products/product of sums are common operations
> 
> That's not really the case in my experience. You're the first person who wanted a first-class sum-of-product operation that I'm aware of.

Minimizing SOP/POS equations with Amaranth can be useful to implement complex instruction decoders. For example, Chisel provides one in its [libraries](https://github.com/chipsalliance/chisel/tree/main/src/main/scala/chisel3/util/experimental/decode), where minimization is done with QMC or by shelling out to Espresso.

This is probably too niche for Amaranth stdlib, but such functionality could make a good downstream library.

jrmoserbaltimore, 2023-11-10:
> Minimizing SOP/POS equations with Amaranth can be useful to implement complex instruction decoders. For example, Chisel provides one in its [...]

Fair point.  I've used similar things when writing a RISC-V decoder or a Gameboy Color MMU emulator, with ugly things like `(x & 'b0101 == 'b0101) | (x & 'b1011 == 'b1010))` (i.e. `(b&d)|(a&c&~d)`) and so forth in verilog long ago.  Lots of overly complex `case` statements I'd rather replace with something more elegant.  ...thinking about that code kind of makes me not feel so bad my SSD died without backups.

Wonder what else would go in such a library, but that's probably a conversation for somewhere other than here.

whitequark, 2024-02-01:
I'm going to close this because there isn't enough interest in the feature that I've seen over the years to continue tracking it. It is still open for commenting, and of course anyone can submit an RFC for this feature or any other one.

closed 2024-02-01 by whitequark