amaranth-lang/amaranth #953: Gowin vendor definitions don't support the GW1NR-LV9QN88PC6/I5 (tang nano 9k) but apicula does

jeremyherbert, 2023-11-15:
According to [this line](https://github.com/YosysHQ/apicula/blob/a4f04e0c481d0ab5feb828a311cca1a76c7b18b6/examples/Makefile#L130) in apicula, the Tang Nano 9k has the device `GW1NR-LV9QN88PC6/I5` and family `GW1N-9C`. However, when using these values with the Gowin vendor support in Amaranth, the following error is emitted from `parse_part`:

```
ValueError: Series extracted from supplied part name does not match supplied family series
```

The apicula readme also has this note at the bottom: 

"C devices require passing the --family flag as well as --device to Nextpnr, and stating the family in place of device when passing -d to gowin_pack because the C isn't part of the device ID but only present in the date code."

jeremyherbert, 2023-11-15:
Also same issue with `GW2AR-LV18QN88C8/I7` and `GW2A-18C` for the tang nano 20k. Also see https://github.com/YosysHQ/apicula/issues/206

whitequark, 2023-11-15:
cc @bl0x who contributed Gowin support--I do not understand how the part numbering works to be honest

bl0x, 2024-07-20:
@jeremyherbert The family given in the Apicula Makefile is **not** the Gowin device family that the device `GW1NR-LV9QN88PC6/I5` belongs to.

The correct family name is `GW1NR-9C`, as stated in the `device_info.csv` file distributed with the Gowin IDE.

Nextpnr generates chipdb files only for family GW1N, but the same file seems to work also with the GW1NR family. This is what is also mentioned in https://github.com/YosysHQ/apicula/issues/206. 

When using Amaranth, the correct family name needs to be specified. This is automatically the case when using the platform definitions from the [amaranth-boards](https://github.com/amaranth-lang/amaranth-boards/tree/main/amaranth_boards) repository, so e.g. for the tang nano 9k:
```python
   part          = "GW1NR-LV9QN88PC6/I5"                                      
   family        = "GW1NR-9C"
```
(Taken from [tang_nano_9k.py#L14](https://github.com/amaranth-lang/amaranth-boards/blob/ad5a939b86020c53e0e193620b96ca19d5960192/amaranth_boards/tang_nano_9k.py#L14))

When using amaranth_boards, the error mentioned above does not appear.

The requirement from the apicula readme is automatically satisfied by the `gowin` toolchain build process in amaranth.

I hope this is a bit clearer now, but if any questions remain, please ask!

whitequark, 2024-07-21:
Thanks @bl0x for the clear explanation! Gowin part names are something else...

jeremyherbert, 2024-07-21:
Thank you for following up on this.

closed 2024-07-21 by jeremyherbert