amaranth-lang/amaranth #1001: GTKWave gets confused by `ArrayLayout`.

zyp, 2023-12-13:
I have an `ArrayLayout(32, 2)`, which results in the following entries in the VCD file:
```
$var wire 64 W output__data $end
$var wire 32 X output__data[0] $end
$var wire 32 Y output__data[1] $end
```

The square brackets appears to confuse GTKWave, because these signals show up like this:
[image]

I'm not sure whether having square brackets in names violates any naming rules, and GTKWave should handle it more gracefully in any case, but I'm filing it here because I figure we might want to avoid the issue either way.

whitequark, 2023-12-13:
> I'm not sure whether having square brackets in names violates any naming rules

No. I checked a while ago, and this should in principle be completely fine. GTKWave does some undocumented and sometimes unsound reinterpretation of the names.

wanda-phi, 2024-02-09:
For memories, I handled this by always adding `\` in front of the signal name. Not sure if this is the greatest thing, but seems to work. I think maybe we should do this for all VCD vars with non-trivial paths?

whitequark, 2024-02-09:
Hm, sure!

PR #1099 (2024-02-09): sim: prefix fields with `\`.
Fixes #1001.
commit 05ac3675 (2024-02-09): sim: prefix fields with `\`.

Fixes #1001.
closed 2024-02-09 by whitequark