amaranth-lang/amaranth #1006: Signal reset cannot be set to a const expression

mcclure, 2023-12-30:
Previously, in #911, I noticed that you could not use a Const() as a reset= argument when creating a Signal. The reason why I wanted this was to make a value with a particular shape.

This has been fixed, but today I noticed that you cannot use a Const() with a mathematical operation on it, such as rotate_right(). The reason I want this is that there is no equivalent to rotate_right() for numbers in the Python standard library, only shift.

## Repro

Consider the branch `z_bug_cellular-toy-consts` of [this repo](https://github.com/mcclure/analogue-core-template-amaranth-variants/tree/z_bug_cellular-toy-consts).

On [this line](https://github.com/mcclure/analogue-core-template-amaranth-variants/blob/z_bug_cellular-toy-consts/src/fpga/amaranth_core/embed_amaranth_core/toplevel.py#L34) I say

    clk_reg = Signal(self.ratio, reset=Const((1 << (self.ratio // 2)) - 1, unsigned(self.ratio)).rotate_right(self.ratio//2-1))

At execution time "self.ratio" is equal to 60. So I want to construct the value caused by filling half the bits in a 60-bit register, then right-rotating it a bit.

If you build this with `(cd src/fpga/amaranth_core/ && python3 -m pdm install && python3 -m pdm generate)`, you get

```
TypeError: Value (slice (const 60'd1073741823) 29:60) cannot be converted to an Amaranth constant
...
TypeError: Reset value must be a constant-castable expression, not (cat (slice (const 60'd1073741823) 29:60) (slice (const 60'd1073741823) 0:29))
```

If you remove the `.rotate_right(self.ratio//2-1)` this line works fine, and a few lines later I am able to `print` the reset value of clk_reg:

```
Clocks 1073741823,32767; stb 288230376151711744
```
 
I am testing with amaranth-yosys 0.35 and Python 3.11.6.

## Expected behavior / could this be fixed another way?

At a high level, my "expected behavior" is that "reasonable" constant Amaranth expressions, such as `Const(43).rotate_left(n)`, `Const(43) + 45` etc should be allowed as `reset=` arguments.

It might be that evaluating constness is nontrivial and that this is not an easy feature to implement. In this case, I would expect the constrained set of expressions which *are* allowed to be documented. I would also expect this low-level/practical thing: *I would expect there to be some way to rotate_left and rotate_right numbers and assign them to `reset=` without using libraries other than Amaranth`. There are several ways this could be achieved:

- Make `reset=` accept `Cat` objects, special-cased, since it is easy to confirm that a `Cat` is a tree of only `Cat` and `Const` expressions.
- Make the `.value` property, documented on `Const`, work on constant `Cat` objects. (Right now this gives the error: `AttributeError: 'Cat' object has no attribute 'value'`.) Then I could call `.value` and assign that to `reset=`.
- Offer rotate-left and rotate-right functions that operate on regular Python integers. They can keep the power-of-two functions in `util` company!

For now, I will possibly be either implementing it myself (it is not hard to implement but it is nice to be able to exclude the math functions as sources of bugs, which is harder if the math functions were implemented bespoke for my program…) or using the "bitstring" library.

whitequark, 2023-12-30:
> Signal reset cannot be set to a const expression

This is factually not the case: you *can* use a const-castable expression in `Signal(reset=)`. The [only constant-castable expressions](https://amaranth-lang.org/docs/amaranth/v0.4.0/lang.html#constant-casting) are combinations of `Const` and `Cat` at the moment though, which doesn't include `Slice` that is produced by a rotate operation.

As discussed on IRC, adding slices wouldn't be too difficult, which should resolve your underlying issue of being unable to use rotates.

closed 2023-12-30 by whitequark

mcclure, 2023-12-30:
Thanks y'all

PR #1009 (2023-12-30): hdl.ast: make `Slice` const-castable.
Fixes #1006.
commit 0849e1af (2023-12-30): hdl.ast: make `Slice` const-castable.

Fixes #1006.