amaranth-lang/amaranth #1018: How to specify top module ports?

mobilinkd, 2024-01-06:
I may have missed it somewhere in the documentation, but am I able to explicitly specify ports for the top module generated by Amaranth?  I ask because Gowin apparently uses the top module parameters to infer an interface to the PSRAM/HyperRAM on their chips. To infer the interface to the PSRAM, one needs to specify the following ports in the top module:

```verilog
module top (
    output [1:0] O_psram_ck,       // Magic ports for PSRAM to be inferred
    output [1:0] O_psram_ck_n,
    inout [1:0] IO_psram_rwds,
    inout [15:0] IO_psram_dq,
    output [1:0] O_psram_reset_n,
    output [1:0] O_psram_cs_n
);
```

I have tried using the wiring component, but that does not seem to affect the top module that is generated. I see from the verilog generator examples how to pass a list of ports to the top module. But I don't see an equivalent way to do this when using the platform interface to generate the bitstream.

whitequark, 2024-01-06:
This pattern isn't supported at the moment in the Gowin platform layer. Most toolchains do not require ports to have specific names but rather use some form of constraint assignment, so it was never added to Amaranth. For the time being your best bet is manually using `verilog.convert` and assigning the `.name` attribute for these ports, or simply naming them that in the source code.

mobilinkd, 2024-01-06:
What Gowin is doing is a hack. There's no excuse for doing it this way.

Is it possible to use the Gowin platform layer to build and upload the bitstream from the generated verilog file, or will it be more straight-forward to automate the bitstream generation and upload using an external tcl script?

whitequark, 2024-01-06:
You could probably grab the build directory generated for a normal Gowin platform build and include it into your build process as-is, substituting the autogenerated Verilog file for your own.

whitequark, 2024-01-06:
> What Gowin is doing is a hack. There's no excuse for doing it this way.

It is, and there isn't, but the Amaranth policy is to meet the toolchains where they are rather than holding the proverbial high ground. However, this is a non-trivial change to the platform layer and it's not a high priority for me.

whitequark, 2024-02-09:
@mobilinkd Is there anything else you would like to be done in regards to this issue?

mobilinkd, 2024-02-16:
No. It can be closed.

Kind Regards,

Rob Riggs WX9O
Mobilinkd LLC

closed 2024-02-25 by whitequark