amaranth-lang/amaranth #1019: `Signal(range(0))` raises `SyntaxWarning`

rroohhh, 2024-01-07:
`Signal(range(0))` raises 
```
SyntaxWarning: Reset value 0 equals the non-inclusive end of the signal shape range(0, 0); this is likely an off-by-one error
```
Is this warning intended? 
To me this warning is quite confusing, as amaranth implicitly converts `range(0)` (which would be something like a never type) to `unsigned(0)`, and `Signal(unsigned(0))` does not raise the same warning. 

Some more discussion can be found in the [IRC logs](https://libera.irclog.whitequark.org/amaranth-lang/2024-01-07#1704643458-1704644743;).

whitequark, 2024-01-08:
> Is this warning intended?

It is intended in the sense that I added that warning and it performs to its specification.

> To me this warning is quite confusing

I agree with Wanda that the behavior is not well-defined. Can you share the code?

whitequark, 2024-01-29:
We have [discussed](https://libera.irclog.whitequark.org/amaranth-lang/2024-01-29#1706551109-1706548020;) this issue on the 2024-01-29 meeting. We have decided to allow `Signal(range(0))` without a warning, and also turn all cases of `Signal(range(X, Y, Z), reset=R)` where `R not in range(X, Y, Z)` into hard errors.

PR #1054 (2024-01-30): hdl.ast: change warning on out-of-range reset to an error, improve it.
Fixes #1019.
commit e9299ccd (2024-01-30): hdl.ast: change warning on out-of-range reset to an error, improve it.

Fixes #1019.
closed 2024-01-30 by whitequark