amaranth-lang/amaranth #1020: cannot open build_top.sh: No such file

fischermoseley, 2024-01-07:
Hello! I think #1012 might have broken builds for Icestick/Vivado-based flows, at least on my machine(s). If I do the following on either MacOS Sonoma 14.2.1 or Ubuntu 22.04:

```
cd /tmp
mkdir foo
cd foo
python3 -m venv venv/
source venv/bin/activate
pip install "amaranth[builtin-yosys] @ git+https://github.com/amaranth-lang/amaranth.git"
pip install amaranth_boards@git+https://github.com/amaranth-lang/amaranth-boards
export PATH=$PATH:/tools/oss-cad-suite/bin
python3 -m amaranth_boards.icestick
```

Then I'll get the following after everything has installed:
```
sh: 0: cannot open build_top.sh: No such file
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/tmp/foo/venv/lib/python3.10/site-packages/amaranth_boards/icestick.py", line 53, in <module>
    ICEStickPlatform().build(Blinky(), do_program=True)
  File "/tmp/foo/venv/lib/python3.10/site-packages/amaranth/build/plat.py", line 109, in build
    products = plan.execute_local(build_dir)
  File "/tmp/foo/venv/lib/python3.10/site-packages/amaranth/build/run.py", line 117, in execute_local
    subprocess.check_call(["sh", f"{self.script}.sh"],
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['sh', 'build_top.sh']' returned non-zero exit status 2.
```

I'm not entirely sure why this is the case - on MacOS `sh` appears to be stored at `/bin/sh`, which matches the path in the shebang. On Ubuntu it's stored at `/usr/bin/sh`, but they both produce the same error.

(unrelated: I really like the new logo, it looks great!)

whitequark, 2024-01-08:
> I really like the new logo, it looks great!

Thanks!

> I'm not entirely sure why this is the case - on MacOS `sh` appears to be stored at `/bin/sh`, which matches the path in the shebang. On Ubuntu it's stored at `/usr/bin/sh`, but they both produce the same error.

That's... baffling. I'm going to take a look.

closed 2024-01-09 by whitequark

fischermoseley, 2024-01-09:
All my builds are working again, thanks for the fix!

anuejn, 2024-01-09:
oh woopsie, sorry for breaking builds and thanks for fixing them @whitequark