amaranth-lang/amaranth #1028: Build fails: pdm.backend._vendor.packaging.version.InvalidVersion: Invalid version: '12.4-eol'

yurivict, 2024-01-11:
The build fails:
```
* Building wheel...
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
    main()
  File "/usr/local/lib/python3.9/site-packages/pyproject_hooks/_in_process/_in_process.py", line 335, in main
    json_out['return_val'] = hook(**hook_input['kwargs'])
  File "/usr/local/lib/python3.9/site-packages/pyproject_hooks/_in_process/_in_process.py", line 251, in build_wheel
    return _build_backend().build_wheel(wheel_directory, config_settings,
  File "/usr/local/lib/python3.9/site-packages/pdm/backend/__init__.py", line 54, in build_wheel
    return builder.build(
  File "/usr/local/lib/python3.9/site-packages/pdm/backend/base.py", line 213, in build
    self.initialize(context)
  File "/usr/local/lib/python3.9/site-packages/pdm/backend/base.py", line 180, in initialize
    self.call_hook("pdm_build_initialize", context)
  File "/usr/local/lib/python3.9/site-packages/pdm/backend/base.py", line 138, in call_hook
    getattr(hook, hook_name)(context, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pdm/backend/hooks/version/__init__.py", line 69, in pdm_build_initialize
    metadata["version"] = getattr(self, f"resolve_version_from_{source}")(
  File "/usr/local/lib/python3.9/site-packages/pdm/backend/hooks/version/__init__.py", line 98, in resolve_version_from_scm
    version = get_version_from_scm(context.root, tag_regex=tag_regex)
  File "/usr/local/lib/python3.9/site-packages/pdm/backend/hooks/version/scm.py", line 333, in get_version_from_scm
    version = func(root, config)  # type: ignore
  File "/usr/local/lib/python3.9/site-packages/pdm/backend/hooks/version/scm.py", line 200, in git_parse_version
    return meta(config, tag, number or None, dirty, node, branch)
  File "/usr/local/lib/python3.9/site-packages/pdm/backend/hooks/version/scm.py", line 80, in meta
    tag = tag_to_version(config, tag)
  File "/usr/local/lib/python3.9/site-packages/pdm/backend/hooks/version/scm.py", line 163, in tag_to_version
    return Version(version)
  File "/usr/local/lib/python3.9/site-packages/pdm/backend/_vendor/packaging/version.py", line 197, in __init__
    raise InvalidVersion(f"Invalid version: '{version}'")
pdm.backend._vendor.packaging.version.InvalidVersion: Invalid version: '12.4-eol'

ERROR Backend subprocess exited when trying to invoke build_wheel
*** Error code 1
```

I suspect that this is because it is built from the tarball, not from the Git repository.

Version: 0.4.0

whitequark, 2024-01-11:
Builds from tarballs are not supported [anymore](https://github.com/amaranth-lang/amaranth/pull/906). If you are installing 0.4.0, why not grab the wheel from PyPI?

yurivict, 2024-01-11:
> If you are installing 0.4.0, why not grab the wheel from PyPI?

Because the package should be built from sources. All packages are built from sources.

whitequark, 2024-01-11:
You can set `PDM_BUILD_SCM_VERSION=0.4.0` to build from a tarball.

yurivict, 2024-01-11:
This works, thanks!

closed 2024-01-11 by yurivict