amaranth-lang/amaranth #1044: Indexing a `Value` with another `Value` should hint at `bit_select` (or just be made to work)

wanda-phi, 2024-01-18:


whitequark, 2024-01-18:
One of the reasons it isn't implemented is that if `x[i]` works you expect `x[i:i+1]` to work, but the latter is something we cannot enable.

zyp, 2024-01-18:
We *could* check whether the slice expression matches the form `[a:a+b]` where `a` is a `Value` and `b` is a `Const` and rewrite it to a `bit_select`, e.g:

```pycon
>>> x = Signal(16)
>>> i = Signal(4)
>>> s = slice(i, i + 1)
>>> s
slice((sig i), (+ (sig i) (const 1'd1)), None)
>>> assert isinstance(s.stop, Operator)
>>> assert s.stop.operator == '+'
>>> assert s.start is s.stop.operands[0]
>>> assert isinstance(s.stop.operands[1], Const)
>>> x.bit_select(s.start, s.stop.operands[1].value)
(part (sig x) (sig i) 1 1)
```

whitequark, 2024-01-18:
> We _could_ check whether the slice expression matches the form `[a:a+b]` where `a` is a `Value` and `b` is a `Const` and rewrite it to a `bit_select`, e.g:

Obviously. But I do not think we should, for the same reason we e.g. I voted against https://github.com/amaranth-lang/rfcs/pull/7. It creates surprising cliffs and makes the language behave less predictably.

whitequark, 2024-01-18:
Or to answer in more detail: OK, we recognize `[a:a+b]`. Why not `[a:b+a]`? Why not `[a:a+b+1]`? Where does it end? There is no reasonable answer you can make that is clear to developers at all times besides "we don't do that, ever, and we have a different syntax, properly statically typed, for this operation.

zyp, 2024-01-18:
I'm inclined to agree, but for the sake of discussion: The end would be any expression that through commutativity is logically equivalent to `[a:a+b]` where `a` is a `Value` and `b` is const-castable with a positive value. Looking through nested addition (and probably subtraction) operators for the `a` operand and const-casting the rest should do it.

Then of course, once we have that, the next expectation would be `word_select` expressions…

whitequark, 2024-01-18:
That's roughly what Verilog does, yes. Amaranth is intentionally not doing that.

PR #1045 (2024-01-18): hdl.ast: suggest bit_select or word_select when indexing with Value.
Fixes #1044.
commit b40c18fb (2024-01-18): hdl.ast: suggest bit_select or word_select when indexing with Value.

Fixes #1044.
closed 2024-01-18 by whitequark