amaranth-lang/amaranth #1049: Miscompilation of nested ifs

nfbraun, 2024-01-22:
The following module:
```
class Test(Elaboratable):
    def __init__(self, in0, in1, in2):
        self.in0 = in0
        self.in1 = in1
        self.in2 = in2

        self.out = Signal()

    def elaborate(self, platform):
        m = Module()

        with m.If(self.in0):
            with m.If(self.in1):
                m.d.sync += self.out.eq(1)
        with m.Elif(self.in2):
            m.d.sync += self.out.eq(0)

        return m
```
gets turned into the following Verilog code:
```
module \top.U$$1 (in0, in1, in2, out, rst, clk);
  reg \$auto$verilog_backend.cc:2253:dump_module$2  = 0;
  input clk;
  wire clk;
  input in0;
  wire in0;
  input in1;
  wire in1;
  input in2;
  wire in2;
  output out;
  reg out = 1'h0;
  reg \out$next ;
  input rst;
  wire rst;
  always @(posedge clk)
    out <= \out$next ;
  always @* begin
    if (\$auto$verilog_backend.cc:2253:dump_module$2 ) begin end
    \out$next  = out;
    if (in0)
      if (in1)
        \out$next  = 1'h1;
    else if (in2)
      \out$next  = 1'h0;
    if (rst)
      \out$next  = 1'h0;
  end
endmodule
```
which, AFAICS, behaves as if the `Elif` had been indented:
```
with m.If(self.in0):
    with m.If(self.in1):
        m.d.sync += self.out.eq(1)
    with m.Elif(self.in2):
        m.d.sync += self.out.eq(0)
```

This is with Amaranth 0.4.1 and Yosys 0.37 (git sha1 a5c7f69ed).

whitequark, 2024-01-22:
Oh, no. This is a bug I have submitted to Yosys.

whitequark, 2024-01-23:
Upstream fix pending: https://github.com/YosysHQ/yosys/pull/4150.

Amaranth should probably explicitly forbid using Yosys 0.37 because of this bug as well.

commit 65d77f03 (2024-01-24): back.verilog: forbid Yosys version range with dangling else bug.

Fixes #1049.
PR #1051 (2024-01-24): back.verilog: forbid Yosys version range with dangling else bug.
Fixes #1049.

This can be merged once https://github.com/YosysHQ/yosys/pull/4150 is in and the version range is final.
closed 2024-01-24 by whitequark