amaranth-lang/amaranth #1089: Build fails due to missing attribute

polymerizedsage, 2024-02-07:
15b6068 seems to break my builds due to the removal of `_all_toolchain_env_vars` from `build/plat.py`. The attribute is still referenced in several places in the `plat.py` file. The call to `Platform.build()` seems to be the issue.

cr1901, 2024-02-07:
Can also confirm that this broke when I did a `pdm update` for Sentinel. At least [ice40](https://github.com/amaranth-lang/amaranth/blob/9e75962c352bc73f4b848acc8d3f5e8c5ed6d02f/amaranth/vendor/_lattice_ice40.py) is affected:

```sh
William@DESKTOP-3H1DSBV MINGW64 ~/Projects/FPGA/amaranth/sentinel
$ pdm update
� Lock successful
Synchronizing working set with resolved packages: 0 to add, 5 to update, 0 to remove

  ✔ Update pyflakes 3.1.0 -> 3.2.0 successful
  ✔ Update flake8 6.1.0 -> 7.0.0 successful
  ✔ Update pytest 7.4.3 -> 8.0.0 successful
  ✔ Update amaranth-soc 0.1.dev79+g622c7ac -> 0.1.dev83+g6f59eec successful
  ✔ Update amaranth 0.4.1.dev6+g8cd8cdd -> 0.4.1.dev67+g9e75962 successful
Installing the project as an editable package...
  ✔ Update sentinel 0.1.0+editable -> 0.1.0 successful

� All complete!

PDM 2.10.1.dev13+g7f0b66d0.editable is installed, while 2.12.3 is available.
Please run `C:/msys64/mingw64/bin/python.exe -m pdm self update` to upgrade.
Run `pdm config check_update false` to disable the check.

William@DESKTOP-3H1DSBV MINGW64 ~/Projects/FPGA/amaranth/sentinel
$ pdm demo
Traceback (most recent call last):
  File "C:/msys64/home/William/Projects/FPGA/amaranth/sentinel/examples/attosoc.py", line 806, in <module>
    main()
  File "C:/msys64/home/William/Projects/FPGA/amaranth/sentinel/examples/attosoc.py", line 802, in main
    demo(args)
  File "C:/msys64/home/William/Projects/FPGA/amaranth/sentinel/examples/attosoc.py", line 724, in demo
    plan = plat.build(asoc, name="rand" if args.r else "top", do_build=False,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:/msys64/home/William/Projects/FPGA/amaranth/sentinel/.venv/lib/python3.11/site-packages/amaranth/build/plat.py", line 92, in

build
    got_env_var = any(v in os.environ for v in self._all_toolchain_env_vars)
                                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ICEStickPlatform' object has no attribute '_all_toolchain_env_vars'. Did you mean: '_toolchain_env_var'?
```

whitequark, 2024-02-08:
https://github.com/amaranth-lang/amaranth/pull/1090

PR #1090 (2024-02-08): Fix toolchain environment variable check
The bug was introduced in commit 15b6068c.

Fixes #1089.
commit 78b90fba (2024-02-08): build.plat,vendor: fix toolchain environment variable check.

The bug was introduced in commit 15b6068c. A changelog entry was also
missing.

Fixes #1089.
closed 2024-02-08 by whitequark