amaranth-lang/amaranth #1107: PySim VCDWriter is useless with `io.StringIO`

whitequark, 2024-02-11:
It unconditionally calls `close` on the VCD and that destroys the `StringIO` object. Questionable design of the latter, perhaps, but we have to live with it.

Not entirely clear what to do with it. I suggest just special-casing `io.StringIO` for specifically `write_vcd(vcd_file=)`. Though, there could be a similar issue with a `tempfile.TemporaryFile` or `tempfile.SpooledTemporaryFile`, so maybe a `close_vcd_file=True` argument would work better.

This is a minor change that does not require an RFC.

zyp, 2024-02-11:
I would argue that the sanest solution is for it to only close file objects it has opened itself, i.e. when passed a filename.

When the caller passes a file object, whether it's a `StringIO`, a regular file or anything else, leave it to the caller to deal with closing it.

whitequark, 2024-02-11:
> I would argue that the sanest solution is for it to only close file objects it has opened itself, i.e. when passed a filename.

You're completely right.

commit a430c1d4 (2024-02-27): sim.pysim: Only close VCD/GTKW files if we opened them ourselves.

Fixes #1107.
PR #1156 (2024-02-27): sim.pysim: Only close VCD/GTKW files if we opened them ourselves.
Fixes #1107.
commit a7252827 (2024-02-27): sim.pysim: Only close VCD/GTKW files if we opened them ourselves.

Fixes #1107.
closed 2024-02-27 by whitequark