amaranth-lang/amaranth #1150: Driver-Driver conflict causes yosys execution with invalid IL if it is over different Elaboratable instances

ECP5-PCIe, 2024-02-17:
The following code warns that the hierarchy will be flattened:
`DriverConflict: Signal '(sig signal)' is driven from multiple fragments: top, top.test; hierarchy will be flattened`

proceeds with Yosys, yielding a syntax error:
`ERROR: Parser error in line 126: syntax error`

The line in question:
`    connect \D None`

```python
from amaranth import *
from amaranth_boards import versa_ecp5_5g as FPGA
from amaranth.vendor import LatticeECP5Platform
import os
from amaranth.lib.fifo import AsyncFIFOBuffered

class TestElaboratable(Elaboratable):
	def __init__(self):
		self.signal = Signal()

	def elaborate(self, platform) -> Module:
		m = Module()

		m.d.comb += self.signal.eq(0)
		# Uncommenting this causes driver-driver conflict error
		#m.d.sync += self.signal.eq(0)

		return m

class MCVE(Elaboratable):
	def elaborate(self, platform):
		m = Module()

		m.submodules.test = test = TestElaboratable()
		m.d.sync += test.signal.eq(0)

		return m

os.environ["AMARANTH_verbose"] = "Yes"
FPGA.VersaECP55GPlatform().build(MCVE())
```

Presumably this is supposed to cause a driver-driver conflict error (as it does when I tried to further simplify it)

wanda-phi, 2024-02-25:
This doesn't seem to reproduce on current main:

```
amaranth.hdl._ir.DriverConflict: Signal (sig signal) driven from domain None at ('/home/mwk/amaranth/amaranth/hdl/_xfrm.py', 667) and domain <amaranth.hdl._cd.ClockDomain object at 0x7f1b23891f50> at ('/home/mwk/amaranth/amaranth/hdl/_xfrm.py', 667)
```

The error message isn't the best (and improving it is on my list), but the error is definitely thrown. What version were you testing on?

wanda-phi, 2024-02-27:
This message is improved in #1158 to: `amaranth.hdl._ir.DriverConflict: Signal (sig signal) driven from domain comb at ('/home/mwk/amaranth/amaranth/hdl/_xfrm.py', 667) and domain sync at ('/home/mwk/amaranth/amaranth/hdl/_xfrm.py', 667)`

An unrelated upcoming change (removing `AssignmentLegalizer`) will also fix up the wrong filename currently reported.

wanda-phi, 2024-02-28:
In #1173 this changes to: `amaranth.hdl._ir.DriverConflict: Signal (sig signal) driven from domain comb at ('/home/mwk/amaranth/t.py', 14) and domain sync at ('/home/mwk/amaranth/t.py', 25)`

PR #1173 (2024-02-28): hdl._ir: Inline `AssignmentLegalizer` into netlist building.
Fixes #1150.
commit f8e2d26b (2024-02-28): hdl._ir: Inline `AssignmentLegalizer` into netlist building.

Fixes #1150.
closed 2024-02-28 by whitequark