amaranth-lang/amaranth #1160: Normalize diagnostics to use either a backtick or a single quote, not both at random

whitequark, 2024-02-27:
Bikeshed: which of these should we use?

tpwrules, 2024-02-28:
What is an example of each in the code as it stands?

I'd say that backticks are better for code examples in like "`x` is bad. Do `y`". But single quotes are better for values, like "x should be a y, not 'z'"

whitequark, 2024-03-04:
> I'd say that backticks are better for code examples in like "`x` is bad. Do `y`". But single quotes are better for values, like "x should be a y, not 'z'"

I like this distinction!

> What is an example of each in the code as it stands?

Backticks: https://github.com/amaranth-lang/amaranth/blob/31a12c03d19fef601b6515c431c5b686d2ba3702/amaranth/back/rtlil.py#L1007

Single quotes: https://github.com/amaranth-lang/amaranth/blob/31a12c03d19fef601b6515c431c5b686d2ba3702/amaranth/build/plat.py#L48