amaranth-lang/amaranth #1209: Zero-length submodule name breaks things

tpwrules, 2024-03-16:
The following design:

```
from amaranth import *
from amaranth.back import verilog

class Bar(Elaboratable):
    def elaborate(self, platform):
        m = Module()
        m.d.sync += Signal().eq(1)
        return m

class Foo(Elaboratable):
    def elaborate(self, platform):
        m = Module()
        m.submodules[""] = Bar()
        return m

print(verilog.convert(Foo(), ports=[]))
```

gives the following exception:
```
Traceback (most recent call last):
  File "~/test.py", line 16, in <module>
    print(verilog.convert(Foo(), ports=[]))
  File "~/venv/lib/python3.10/site-packages/amaranth/back/verilog.py", line 61, in convert
    verilog_text, name_map = convert_fragment(fragment, ports, name, emit_src=emit_src, strip_internal_attrs=strip_internal_attrs, **kwargs)
  File "~/venv/lib/python3.10/site-packages/amaranth/back/verilog.py", line 39, in convert_fragment
    rtlil_text, name_map = rtlil.convert_fragment(*args, **kwargs)
  File "~/venv/lib/python3.10/site-packages/amaranth/back/rtlil.py", line 1029, in convert_fragment
    empty_checker=empty_checker).emit()
  File "~/venv/lib/python3.10/site-packages/amaranth/back/rtlil.py", line 308, in emit
    self.emit_submodules()
  File "~/venv/lib/python3.10/site-packages/amaranth/back/rtlil.py", line 516, in emit_submodules
    self.builder.cell(f"\\{dotted_name}", submodule.name[-1], ports={
  File "~/venv/lib/python3.10/site-packages/amaranth/back/rtlil.py", line 166, in cell
    name = self._make_name(name, local=False)
  File "~/venv/lib/python3.10/site-packages/amaranth/back/rtlil.py", line 67, in _make_name
    elif not local and name[0] not in "\\$":
IndexError: string index out of range
```

I suggest that `m.submodules[""] = ...` have the same behavior as `m.submodules += ...`. While it would be silly to directly write the former, it might arise from programmatic name generation, and should either work properly or immediately throw an error.

tpwrules, 2024-03-23:
Upon further consideration it's probably best to reject "" as an invalid name rather than try to do something magic with it.

whitequark, 2024-03-24:
> I suggest that `m.submodules[""] = ...` have the same behavior as `m.submodules += ...`. While it would be silly to directly write the former, it might arise from programmatic name generation, and should either work properly or immediately throw an error.

I was actually thinking that this proposal seems OK.

tpwrules, 2024-03-24:
The specific situation I had in mind was that if you do `m.submodules[x] = foo`, then `bar = m.submodules[x]`, this cannot work if `x == ""`.

I'm fine with the concept of unnamed modules, but if you accidentally make `x == ""`, then having the above break might be confusing. It does seem unlikely though.

whitequark, 2024-03-24:
That's a good point, I forgot `m.submodules[x]` (reading) is a thing.

PR #1376 (2024-06-10): Forbid empty string as submodule name
This is semantically ambiguous and breaks the RTLIL emitter.

Fixes #1209.
commit 0140fe27 (2024-06-10): hdl._dsl: forbid empty string as submodule name.

This is semantically ambiguous and breaks the RTLIL emitter.

Fixes #1209.
closed 2024-06-10 by whitequark