amaranth-lang/amaranth #1213: Tracking issue for RFC 36: Async testbench functions

zyp, 2024-03-18:
> * RFC PR: [Add an RFC for async testbench functions. rfcs#36](https://github.com/amaranth-lang/rfcs/pull/36)
> * Implementation PR: #990

whitequark, 2024-05-09:
This still needs to be documented, so I'll keep it open for now.

cr1901, 2024-06-15:
> This still needs to be documented, so I'll keep it open for now.

Will generator functions be deprecated in 0.6 instead of 0.5? `async` testbenches made the 0.5 cutoff, and the RFC mentions deprecation of generator functions at the very end. But I don't see anything about deprecation in the CHANGELOG/porting guide from 0.4 to 0.5.

(Basically, "I'd like to quickly migrate to 0.5 and cut a release of some packages without upgrading generator testbenches to `async` for now.")

whitequark, 2024-06-15:
Generator functions are deprecated in 0.5 and this is mentioned in the porting guide:

> Convert simulator processes and testbenches to use the new async API.

However the deprecation fell through the cracks. (@wanda-phi do you think you can do another changelog backport PR?)