amaranth-lang/amaranth #1221: Add `MemoryData`; like `Signal` but for memories

whitequark, 2024-03-21:
Right now the memory geometry (shape/width and depth) are set either on `MemoryInstance` (the internal perma-unstable primitive) or on `lib.memory.Memory` (the user-facing interface). However, this creates a problem for the simulator:
- `MemoryInstance` is not really usable for most code, so whatever it's doing is irrelevant for 99% of users.
- `lib.memory.Memory` can't be directly read using `sim.memory_{read,write}` from RFC 36; those primitives only work on `MemoryIdentity`, which is a marker class whose only purpose is to have an identity.
  - Also we have no primitives for calling `sim.memory_{read,write}` on `lib.memory.Memory` at the moment.
- Neither `lib.memory.Memory` nor `MemoryInstance` can be passed to `traces=[]`, you have to pass a `MemoryInstance`.

To solve this, I think we should have a `MemoryData(shape: ShapeLike, depth: int, init: MemoryData.Init)` class, which is essentially the 2d counterpart of `Signal()`. Once we add this, we:
- Remove/replace `MemoryIdentity` with `MemoryData`.
- Teach `sim.memory_{read,write}` to use `MemoryData`.
- Teach `write_vcd(traces=[...])` to use `MemoryData`.
- Add a `lib.memory.Memory(data=)` constructor that accepts `MemoryData`.
  - Unclear whether the existing constructor `lib.memory.MemoryData(shape=, depth=)` should be kept.

wanda-phi, 2024-03-22:
RFCd as amaranth-lang/rfcs#62

whitequark, 2024-03-25:
Superseded by #1241.

closed 2024-03-25 by whitequark