amaranth-lang/amaranth #1223: Signals with private names

whitequark, 2024-03-22:
In some cases it is desirable to create a signal with a name that doesn't show up in the waveform viewer, such as when a signal is created to hold the results of intermediate computations, or to change shape, and it would just add noise. I propose adding a notion of a "private signal name".

I suggest changing naming rules so that `Signal(name="")` creates a signal with a private name. Such a signal is translated to a private wire in RTLIL (with a `$\d+` name), doesn't get displayed in the waveform viewer, and so on.

---

The FSM ongoing signals (not the state signal) should be changed to have private names in the 0.5 milestone.

tpwrules, 2024-03-22:
Would this ever be the default for a signal generated by Amaranth, or would a user always have to create it explicitly?

whitequark, 2024-03-22:
> Would this also be the default for some signals generated by Amaranth

Only the ones that are implementation details (currently just `with m.FSM()` but I think we might have others later).

tpwrules, 2024-03-22:
I think this should be straightforward, I will try to cook up an implementation this weekend and update by Monday.

PR #1234 (2024-03-25): Allow Signals to be privately named using `name=""`
Closes #1223

Tested a couple toy designs. The inclusion of FSM names being private should be enough to prevent regressions.

This does remove a bit of diagnostic power from the FSM tests.
closed 2024-03-25 by whitequark