amaranth-lang/amaranth #1246: `execute_local_docker()` does not forward SIGINT/ctrl+c to docker subprocess

polymerizedsage, 2024-03-25:
I use the `execute_local_docker()` build command almost exclusively in my current project. I have noticed that if I ctrl+c a build the docker subprocess that is spawned by Amaranth doesnt exit. This is undesirable because it can lead to mistakenly spawning two docker containers which are both actively modifying the build directory.

whitequark, 2024-03-25:
Ah yeah that's unpleasant. Want to fix it? It will require catching `KeyboardInterrupt`, I think.

polymerizedsage, 2024-03-25:
Yep should be a pretty simple fix. Ill PR when I get a chance, probably tomorrow.

whitequark, 2024-04-09:
@polymerizedsage Have you been able to fix this issue?

polymerizedsage, 2024-04-09:
gonna pretend I didnt forget about this and instead have just been lazy on opening the PR. Heres the fix:
`subprocess.check_call()` should handle catching the `KeyboardInterrupt` and sending SIGKILL/SIGTERM but that doesnt seem to work well for docker container processes. Having the container run an interactive shell with `-it` instead means the command running in the container will properly receive the ctrl-c input.

PR #1296 (2024-04-09): run: fix `execute_local_docker()` not properly catching ctrl-c
fix #1246.
closed 2024-04-09 by whitequark