amaranth-lang/amaranth #1260: Wrong verilog for combinatorial module

FatsieFS, 2024-04-02:
I have the following component:
```python
class RGBOn(wiring.Component):
    def __init__(self, *, r_shape, g_shape, b_shape):
        super().__init__({
            "halfbrite": In(1),

            "r_on": Out(r_shape),
            "b_on": Out(b_shape),
            "g_on": Out(g_shape),
        })

    def elaborate(self, platform):
        m = Module()

        m.d.comb += (
            self.r_on.eq(~0),
            self.g_on.eq(~0),
            self.b_on.eq(~0),
        )
        with m.If(self.halfbrite):
            m.d.comb += (
                self.r_on[-1].eq(0),
                self.g_on[-1].eq(0),
                self.b_on[-1].eq(0),
            )

        return m
```

It generates to following verilog output:
```verilog
module \top.U$1.linegen.rgb_on (g_on, b_on, halfbrite, r_on);
  reg \$auto$verilog_backend.cc:2341:dump_module$3  = 0;
  output [5:0] b_on;
  reg [5:0] b_on;
  output [5:0] g_on;
  reg [5:0] g_on;
  input halfbrite;
  wire halfbrite;
  output [5:0] r_on;
  reg [5:0] r_on;
  always @* begin
    if (\$auto$verilog_backend.cc:2341:dump_module$3 ) begin end
    r_on[5] = 1'h1;
    if (halfbrite) begin
      r_on[5] = 1'h0;
    end
  end
  always @* begin
    if (\$auto$verilog_backend.cc:2341:dump_module$3 ) begin end
    g_on[5] = 1'h1;
    if (halfbrite) begin
      g_on[5] = 1'h0;
    end
  end
  always @* begin
    if (\$auto$verilog_backend.cc:2341:dump_module$3 ) begin end
    b_on[5] = 1'h1;
    if (halfbrite) begin
      b_on[5] = 1'h0;
    end
  end
  assign b_on[4:0] = 5'h1f;
  assign g_on[4:0] = 5'h1f;
  assign r_on[4:0] = 5'h1f;
endmodule
```

This fails in Quartus 13.1 with the following errors for the last three `assign` lines:
```console
Error (10219): Verilog HDL Continuous Assignment error at top.v(387): object "b_on" on left-hand side of assignment must have a net type File: /home/verhaegs/eda/code/amaranth-boards/build/top.v Line: 387
Error (10219): Verilog HDL Continuous Assignment error at top.v(388): object "g_on" on left-hand side of assignment must have a net type File: /home/verhaegs/eda/code/amaranth-boards/build/top.v Line: 388
Error (10219): Verilog HDL Continuous Assignment error at top.v(389): object "r_on" on left-hand side of assignment must have a net type File: /home/verhaegs/eda/code/amaranth-boards/build/top.v Line: 389
```
Also Vivado errors on these lines so generated Verilog seems to be wrong, I believe assign is not valid for a `reg`.  
This happens with both with Amaranth 0.4.4 and current `main` branch. `amaranth-yosys` is version 0.39.0.149.post89

wanda-phi, 2024-04-02:
Duplicate of #717

whitequark, 2024-04-03:
Will be fixed by https://github.com/YosysHQ/yosys/pull/4314, once it's available for use with Amaranth (likely by way of an amaranth-yosys bump).

closed 2024-04-03 by whitequark