amaranth-lang/amaranth #1361: Extremely Verilog generation

resistor, 2024-05-18:
I'm seeing very slow Verilog generation (at least 10s of minutes) for the following relatively straightforward module. It simulates successfully in Amaranth, but something in the Verilog generation process is going awry.

No claim is made that this is *good* code, but it probably shouldn't take forever to convert to Verilog.

```
from amaranth import *
from amaranth.hdl import *
from amaranth.back import verilog

class VectorPerm(Elaboratable):
    def __init__(self, register_width: int):
        self._register_width = register_width
        self.in_lhs = Signal(unsigned(register_width))
        self.in_rhs = Signal(unsigned(register_width))
        self.in_control = Signal(unsigned(register_width))
        self.out_res = Signal(unsigned(register_width))
    
    def elaborate(self, platform):
        m = Module()

        lhs_list = [self.in_lhs[i*8:(i+1)*8] for i in range(self._register_width // 8)]
        rhs_list = [self.in_rhs[i*8:(i+1)*8] for i in range(self._register_width // 8)]
        lists = lhs_list + rhs_list
        parts_array = Array(lists)
        
        control_list = [self.in_control[i*8:(i+1)*8] for i in range(self._register_width // 8)]

        selects = [parts_array[control_list[i]] for i in range(self._register_width // 8)]
        m.d.comb += self.out_res.eq(Cat(*selects))

        return m

if __name__ == '__main__':
    vec_perm = VectorPerm(register_width=128)
    with open('vec_perm.v', 'w') as f:
        f.write(verilog.convert(vec_perm,
                                ports = [
                                            vec_perm.in_lhs,
                                            vec_perm.in_rhs,
                                            vec_perm.in_control,
                                            vec_perm.out_res,
                                        ]))
```

whitequark, 2024-05-18:
Which version of Amaranth is this?

resistor, 2024-05-18:
> Which version of Amaranth is this?

0.4.5

whitequark, 2024-05-18:
Try HEAD. We have completely re-engineered the backend to avoid issues like this.

resistor, 2024-05-18:
Confirmed that it completes almost instantly on HEAD

whitequark, 2024-05-18:
Closing as this won't matter after the next release.

closed 2024-05-18 by whitequark