amaranth-lang/amaranth #1368: Adding simulator processes after starting the simulation is not well-defined

whitequark, 2024-05-30:
... and should probably be rejected with a diagnostic.

I realized that we technically permit this while writing https://github.com/amaranth-lang/rfcs/pull/67#issuecomment-2139178160, and I don't think we should; among other things it breaks the semantics of `reset` (which rolls back the simulation to the point before `run()`).

zyp, 2024-05-30:
I agree, and I figure that for any case where creating processes during simulation is actually useful, we could eventually add a construct analogous to `asyncio.TaskGroup` or `trio.Nursery` that allows creating multiple child processes with a well defined lifetime.

One use case for this would be the ability to write a function that launces one child process that feeds input into one end of a stream pipeline and another child process that checks the output from the other end.

whitequark, 2024-05-30:
I agree that there are clear and valuable use cases. It is just the lifetime issue that causes concern for me.

commit 6a2e7893 (2024-06-10): sim: forbid adding stimuli to a running simulation.

Fixes #1368.
PR #1378 (2024-06-10): Forbid adding stimuli to a running simulation
Fixes #1368.
closed 2024-06-10 by whitequark