amaranth-lang/amaranth #1404: `stream.Signature` should allow setting the init value for the payload

whitequark, 2024-06-17:
This for example does not work:

```python
class BitSerializer(wiring.Component):
    def __init__(self, *, width, length):
        self._length = length

        super().__init__({
            "shift": In(1),
            "stream": In(stream.Signature(data.StructLayout({
                "o":  data.ArrayLayout(width, length),
                "oe": 1
            })), init={"payload": {"oe": 1}}),
            "buffer": Out(io.FFBuffer.Signature("o", width)),
        })
```

but could be valuable for the same reason `io.Buffer.Signature`'s `oe` member has `init=1`.

wanda-phi, 2024-06-18:
doing it with the proposed syntax is not possible with current `lib.wiring` (a member of signature type with `init` will be rejected)

what do you think about `stream.Signature(..., payload_init={"oe": 1})` instead?

whitequark, 2024-06-29:
Let's bikeshed `payload_init=` vs `init=` on the next meeting.

whitequark, 2024-07-01:
We have [discussed](https://libera.irclog.whitequark.org/amaranth-lang/2024-07-01#1719852888-1719853716;) this issue on the 2024-07-01 weekly meeting. We have decided to go with `payload_init=`.

PR #1434 (2024-07-12): Add `payload_init=` argument to `lib.stream.Signature()`
Fixes #1404.
commit 7b12ec34 (2024-07-12): lib.stream: add `payload_init=` argument to `Signature()`.

Fixes #1404.
commit d0b01dd1 (2024-07-12): lib.stream: add `payload_init=` argument to `Signature()`.

Fixes #1404.
closed 2024-07-12 by whitequark