amaranth-lang/amaranth #1506: Driving ResetSignal() results in multiple drivers error

daniestevez, 2024-09-07:
In Amaranth v0.4.5 and below it was possible to generate Verilog code corresponding to the following snippet:

```python
import amaranth.cli
from amaranth import *

class TestReset(Elaboratable):
    def __init__(self):
        self.a = Signal()
        self.b = Signal()

    def elaborate(self, platform):
        m = Module()
        m.d.comb += ResetSignal().eq(self.a)
        m.d.sync += self.b.eq(~self.b)
        return m

if __name__ == '__main__':
    top = TestReset()
    amaranth.cli.main(top, ports=[top.a, top.b])
```

However, this fails in Amaranth v0.5.0 and above with the following error
```
amaranth.hdl._ir.DriverConflict: Bit 0 of signal (sig rst) has multiple drivers: test_reset.py:11 and /usr/local/lib/python3.12/dist-packages/amaranth/hdl/_ir.py:283
```

[Link to the playground where this can be tested](https://amaranth-lang.org/play/#eyJhdiI6InYwLjUuMCIsInMiOiJmcm9tIGFtYXJhbnRoIGltcG9ydCAqXG5mcm9tIGFtYXJhbnRoLnNpbSBpbXBvcnQgU2ltdWxhdG9yLCBUaWNrXG5mcm9tIGFtYXJhbnRoLmJhY2sgaW1wb3J0IHJ0bGlsLCB2ZXJpbG9nXG5pbXBvcnQgYW1hcmFudGhfcGxheWdyb3VuZFxuXG5cbmNsYXNzIFRlc3RSZXNldChFbGFib3JhdGFibGUpOlxuICAgIGRlZiBfX2luaXRfXyhzZWxmKTpcbiAgICAgICAgc2VsZi5hID0gU2lnbmFsKClcbiAgICAgICAgc2VsZi5iID0gU2lnbmFsKClcblxuICAgIGRlZiBlbGFib3JhdGUoc2VsZiwgcGxhdGZvcm0pOlxuICAgICAgICBtID0gTW9kdWxlKClcbiAgICAgICAgbS5kLmNvbWIgKz0gUmVzZXRTaWduYWwoKS5lcShzZWxmLmEpXG4gICAgICAgIG0uZC5zeW5jICs9IHNlbGYuYi5lcSh+c2VsZi5iKVxuICAgICAgICByZXR1cm4gbVxuXG50b3AgPSBUZXN0UmVzZXQoKVxuXG5hbWFyYW50aF9wbGF5Z3JvdW5kLnNob3dfdmVyaWxvZyh2ZXJpbG9nLmNvbnZlcnQodG9wLCBwb3J0cz1bdG9wLmEsIHRvcC5iXSkpXG4ifQ==)

In my case, the use case I have for this particular kind of construction is to drive the reset of most of the design (which is on the `sync` clock domain) from a register that lives on a different clock domain. I have been taking a look at the code to try to find what else might be driving reset and I haven't found anything.

whitequark, 2024-09-07:
I think this happens because `cd_sync.rst` is a toplevel input and you can't override the values of toplevel inputs (they are considered driven by external logic). I would say that the error is correct, if inconvenient. I haven't examined the netlist generated by Amaranth 0.4.5 yet but I'm uncertain whether it's even well-formed; driver conflicts across hierarchy weren't detected in Amaranth 0.4.

The solution to this problem is to create a clock domain manually.

whitequark, 2024-09-07:
Yeah, the code is definitely wrong. The netlist looks like this:

```verilog
/* Generated by Yosys 0.44 (git sha1 80ba43d26, ccache clang 14.0.0-1ubuntu1.1 -O3 -flto -flto) */

module top(b, clk, rst, a);
  reg \$auto$verilog_backend.cc:2352:dump_module$1  = 0;
  wire \$1 ;
  input a;
  wire a;
  output b;
  reg b = 1'h0;
  reg \b$next ;
  input clk;
  wire clk;
  output rst;
  wire rst;
  assign \$1  = ~ b;
  always @(posedge clk)
    b <= \b$next ;
  always @* begin
    \b$next  = \$1 ;
    if (rst) begin
      \b$next  = 1'h0;
    end
  end
  assign rst = a;
endmodule
```

Having `rst` flip from input to output in this manner is definitely not a behavior that I ever considered for the bit of logic that automatically adds clk/rst on the toplevel; those are supposed to always be inputs.

daniestevez, 2024-09-07:
> The solution to this problem is to create a clock domain manually.

That seems a reasonable solution, but in that case I have no use for the `clk` and `rst` ports of the default `sync` clock domain, since nothing will be using this clock domain. Can I somehow remove them from the generated Verilog `module`?

whitequark, 2024-09-07:
If you call your own domain `sync` the automatic behavior will be suppressed.

daniestevez, 2024-09-07:
I didn't know that. Awesome. That works great! Closing this issue.

closed 2024-09-07 by daniestevez