amaranth-lang/amaranth #1508: Broken Trigger when for TriggerCombination delay/changed expires at the same time

purdeaandrei, 2024-09-08:
When awaiting a TriggerCombination that waits for both a delay() to expire and a signal to change, if the delay expires at the same time as when the signal changes, BrokenTrigger is throws.
Here is minimal example code:

```python
from amaranth import *
from amaranth.sim import Simulator

sig = Signal()

class Dut(Elaboratable):
    def elaborate(self, platform):
        m = Module()
        return m

dut = Dut()

async def bench_a(ctx):
    await ctx.delay(1e-6)
    await ctx.delay(0.5e-6).changed(sig)

async def bench_b(ctx):
    await ctx.delay(1e-6)
    await ctx.delay(0.5e-6)
    ctx.set(sig, 1)

sim = Simulator(dut)
sim.add_testbench(bench_b)
sim.add_testbench(bench_a)

with sim.write_vcd("t.vcd"):
    sim.run()
```

purdeaandrei, 2024-09-09:
Looks like BrokenTrigger can also happen with a single `.changed()` trigger that lists more than one signal.

whitequark, 2024-09-09:
That sounds plausible, yeah.