amaranth-lang/amaranth #1522: Memory initialization makes exported Verilog code very long

epkRichi, 2024-09-28:
I noticed that when coverted to Verilog code, the `amaranth.lib.memory.Memory` class creates a Verilog array where each element gets initialized in its own line. So a memory with a depth of 8 will create this Verilog code:
```
initial begin
data_memory[0] = 32'd0;
data_memory[1] = 32'd0;
data_memory[2] = 32'd0;
data_memory[3] = 32'd0;
data_memory[4] = 32'd0;
data_memory[5] = 32'd0;
data_memory[6] = 32'd0;
data_memory[7] = 32'd0;
end
```

And if the memory has a depth of a million, well, there will be a million lines just for initialization, which is a problem since generating the code takes very long and the file also gets very big. Is there a way to prevent those lines from getting generated?

I have read https://github.com/amaranth-lang/amaranth/issues/270, but that issue discusses the differences of resets on FPGAs and ASICs. If array initialization in the Verilog code is needed, couldn't that be done with a loop instead of generating one line per element?

Amaranth version: 0.5.0

Example code using a Memory:
```
from amaranth import *
from amaranth.lib.memory import Memory
from amaranth.lib import wiring
from amaranth.lib.wiring import In, Out
from amaranth.build import Platform
from amaranth.back import verilog

class MyMemory(wiring.Component):
    def __init__(self, address_width: int, data_width: int):
        self.address_width = address_width
        self.data_width = data_width

        super().__init__(
            {
                "address_i": In(self.address_width),
                "write_data_i": In(self.data_width),
                "write_strobe_i": In(data_width // 8),
                "request_valid_i": In(1),
                "read_data_o": Out(data_width),
            }
        )

    def elaborate(self, platform: Platform):
        m = Module()

        m.submodules.data_memory = data_memory = Memory(
            shape=unsigned(self.data_width), depth=(2**self.address_width), init=[]
        )

        # write port configuration
        write_port = data_memory.write_port(granularity=8)
        m.d.comb += write_port.addr.eq(self.address_i)
        m.d.comb += write_port.data.eq(self.write_data_i)
        m.d.comb += write_port.en.eq(Mux(self.request_valid_i, self.write_strobe_i, 0))

        # read port configuration
        read_port = data_memory.read_port()
        m.d.comb += read_port.addr.eq(self.address_i)
        m.d.comb += read_port.en.eq(self.request_valid_i & ~self.write_strobe_i.any())
        m.d.comb += self.read_data_o.eq(read_port.data)

        return m

top = MyMemory(address_width=3, data_width=32)
with open("my_memory.v", "w") as f:
    f.write(verilog.convert(top, emit_src=False))
```

whitequark, 2024-09-28:
> Is there a way to prevent those lines from getting generated?

No. This is a limitation of Yosys. At the time there are no specific plans to work on it.

epkRichi, 2024-09-29:
But if all elements should get initialized to the same value, cant that be done by using a loop like this:

```
int i;
for (i = 0; i < LEN; i = i + 1) begin
  data_memory[i] = 0;
end
```

Or does that also not work with Yosys?

Of course that only works if the entire memory should get initialized to the same value.

whitequark, 2024-09-29:
> Or does that also not work with Yosys?

Amaranth does not output Verilog. It outputs RTLIL, and asks Yosys to produce Verilog out of it. Yosys always writes out memory contents one-by-one.

epkRichi, 2024-09-29:
Okay, got it.

closed 2024-09-29 by epkRichi

whitequark, 2024-09-29:
I've actually put in some work over the years trying to get Yosys to output the initialization data via `$readmemb`, but it doesn't work when integrated with Amaranth. If you output a `.il` file from Amaranth and then use Yosys manually like `yosys foo.il -b 'write_verilog -extmem' -o foo.v`, you should get a significant speedup.