amaranth-lang/amaranth #1523: Creating Memory Arrays without creating another submodule

epkRichi, 2024-10-02:
In Verilog, you can create an array like this: `reg[31:0] my_array[1023:0]`. You can then directly read from or write to the array. The array can also be synthesized as BRAM if you don't access it incorrectly. 

In Amaranth, the only way I've found to create an array that can be synthesized as BRAM is by using `lib.memory.Memory`. But my problem with that is that this Memory has to be instantiated as a submodule and you have to communicate with it through read and write ports. If you want the memory to be synthesized as BRAM, the read port has to be synchronous. So if you want to access some data from the memory, you've got to initiate a read transfer by setting the signals on the read port accordingly, but the read data will only be available two cycles later.

So reading from `lib.memory.Memory` takes two additional cycles compared to what I would do in Verilog, plus you have to enable and disable the read port.

I've also tried using `data.ArrayLayout` and `hdl.Array`, but they serve other purposes and they get translated to one very big register or one reg per element in the verilog code which cant be synthesized to BRAM as far as I know.

Is this a legitimate problem or am I just trying to do something you're not supposed to do? (I don't have a ton of experience designing hardware, so that might very well be the case)

whitequark, 2024-10-02:
> The array can also be synthesized as BRAM if you don't access it incorrectly.

Correct. This is an important part of why Amaranth memories are designed the way they are: the access pattern shouldn't cause "spooky action at a distance", influencing the synthesized logic for a memory array.

Or if you think of it from a different perspective: Verilog arrays are meant to be *just* arrays, same as in any other programming language, while Amaranth arrays are meant to be specifically *memory* arrays, whose interface and representation are optimized for memory-like access patterns.

> If you want the memory to be synthesized as BRAM, the read port has to be synchronous.

This is also true for Verilog. The only difference is that the access pattern is implicit in Verilog (a part of the containing scope), rather than explicit (a parameter to `.read_port()`).

> So if you want to access some data from the memory, you've got to initiate a read transfer by setting the signals on the read port accordingly, but the read data will only be available two cycles later.

Yes, but it should be one cycle later. I'm not sure how you end up with two; can you provide an example, maybe in the [Amaranth Playground](https://play.amaranth-lang.org/)?

This will also show you a Verilog equivalent for the code you're writing, which might make it a bit easier to correlate the two.

> So reading from `lib.memory.Memory` takes two additional cycles compared to what I would do in Verilog,

You will never see a Verilog array with asynchronous read access be (correctly) synthesized as BRAM, because BRAMs require reads to be clocked on every FPGA family I'm aware of. This is a property of the underlying primitive, not a limitation of the language. (It seems that you're aware of it, but I will also note that Amaranth supports asynchronous read ports, which synthesize to LUTRAM or FFRAM).

> plus you have to enable and disable the read port.

The initial value for the `en` signal of the read port is 1, so you can leave it out. (There are some reasons to access BRAMs in a read-xor-write pattern, but it seemed fine to ignore them for the default.)

epkRichi, 2024-10-02:
Thanks for the quick reply. I've created a small example in the playground [here](https://amaranth-lang.org/play/#eyJhdiI6InYwLjUuMiIsInMiOiJmcm9tIGFtYXJhbnRoIGltcG9ydCAqXG5mcm9tIGFtYXJhbnRoLnNpbSBpbXBvcnQgU2ltdWxhdG9yXG5mcm9tIGFtYXJhbnRoLmJhY2sgaW1wb3J0IHJ0bGlsLCB2ZXJpbG9nXG5mcm9tIGFtYXJhbnRoLmxpYi5tZW1vcnkgaW1wb3J0IE1lbW9yeVxuZnJvbSBhbWFyYW50aC5saWIgaW1wb3J0IHdpcmluZ1xuZnJvbSBhbWFyYW50aC5saWIud2lyaW5nIGltcG9ydCBJbiwgT3V0XG5pbXBvcnQgYW1hcmFudGhfcGxheWdyb3VuZFxuXG5tID0gTW9kdWxlKClcbm0uc3VibW9kdWxlcy5teW1lbW9yeSA9IG15bWVtb3J5ID0gTWVtb3J5KHNoYXBlPXVuc2lnbmVkKDMyKSwgZGVwdGg9OCwgaW5pdD1baSBmb3IgaSBpbiByYW5nZSg4KV0pXG5cbnJlYWRfcG9ydCA9IG15bWVtb3J5LnJlYWRfcG9ydChkb21haW49XCJzeW5jXCIpXG53cml0ZV9wb3J0ID0gbXltZW1vcnkud3JpdGVfcG9ydChkb21haW49XCJzeW5jXCIpXG5cbnN0YXRlID0gU2lnbmFsKHVuc2lnbmVkKDIpKVxuY29tcHV0YXRpb24gPSBTaWduYWwodW5zaWduZWQoMzIpKVxuXG53aXRoIG0uSWYoc3RhdGUgPT0gMCk6XG4gICAgbS5kLnN5bmMgKz0gcmVhZF9wb3J0LmFkZHIuZXEoNSlcbiAgICBtLmQuc3luYyArPSByZWFkX3BvcnQuZW4uZXEoMSlcbiAgICBtLmQuc3luYyArPSBzdGF0ZS5lcSgxKVxuICAgIG0uZC5zeW5jICs9IFByaW50KFwiU3RhdGUgMCAtIHJlYWQgZGF0YTogXCIsIHJlYWRfcG9ydC5kYXRhKVxud2l0aCBtLkVsaWYoc3RhdGUgPT0gMSk6XG4gICAgbS5kLnN5bmMgKz0gcmVhZF9wb3J0LmVuLmVxKDApXG4gICAgbS5kLnN5bmMgKz0gc3RhdGUuZXEoMilcbiAgICBtLmQuc3luYyArPSBQcmludChcIlN0YXRlIDEgLSByZWFkIGRhdGE6IFwiLCByZWFkX3BvcnQuZGF0YSlcbndpdGggbS5FbGlmKHN0YXRlID09IDIpOlxuICAgICMgZGF0YSBjYW4gb25seSBiZSB1c2VkIG5vdyB0byBjb21wdXRlIHRoaW5ncywgbGlrZSBmb3IgZXhhbXBsZTpcbiAgICBtLmQuc3luYyArPSBjb21wdXRhdGlvbi5lcShyZWFkX3BvcnQuZGF0YSAqIDMgKyAyKVxuICAgICMgb3IgdG8gZG8gYnJhbmNoaW5nIGxpa2UgdGhpczpcbiAgICAjIHdpdGggbS5JZihyZWFkX3BvcnQuZGF0YSA9PSAuLi4pOlxuICAgIG0uZC5zeW5jICs9IFByaW50KFwiU3RhdGUgMiAtIHJlYWQgZGF0YTogXCIsIHJlYWRfcG9ydC5kYXRhKVxuXG5hc3luYyBkZWYgYmVuY2goY3R4KTpcbiAgICBhd2FpdCBjdHgudGljaygpLnJlcGVhdCgzKVxuXG5zaW0gPSBTaW11bGF0b3IobSlcbnNpbS5hZGRfY2xvY2soMWUtNilcbnNpbS5hZGRfdGVzdGJlbmNoKGJlbmNoKVxuI3dpdGggYW1hcmFudGhfcGxheWdyb3VuZC5zaG93X3dhdmVmb3JtcyhzaW0pOlxuc2ltLnJ1bigpXG5cbmFtYXJhbnRoX3BsYXlncm91bmQuc2hvd192ZXJpbG9nKHZlcmlsb2cuY29udmVydChtLCBwb3J0cz1bXSkpXG4ifQ==)

The example contains a small state machine where in state 0, I decide that I want to do some computation or maybe even branch based on the value of the memory at some address (which is always 5 in this example). So in state 0, I send a request via the read port. Since the read port is synchronous, the memory will only "start working" on that request in the next cycle, meaning that I can only access the read data two cycles later - in state 2. But what I'd really like to do is to already use the read data in state 0. In Verilog, I could do that:

```
reg[31:0] mymemory[7:0];
reg[1:0] state;
reg[31:0] computation;

always @ (posedge clk) begin
    if (state == 0) begin
        computation <= mymemory[5] * 3 + 2;
        if (mymemory[5] == ...) begin ...
    end
end
```

(this is only an incomplete snippet)

> The initial value for the en signal of the read port is 1, so you can leave it out. (There are some reasons to access BRAMs in a read-xor-write pattern, but it seemed fine to ignore them for the default.)

So it is fine to always leave the read port enabled and only change the address when I want to access a new value, even for the synthesized design?

Btw. the playground example I shared throws an error if i record the waveform (line 40). Not sure if that is a problem with my code or with the playground.

whitequark, 2024-10-02:
> So it is fine to always leave the read port enabled and only change the address when I want to access a new value, even for the synthesized design?

Yes.

> Since the read port is synchronous, the memory will only "start working" on that request in the next cycle, meaning that I can only access the read data two cycles later - in state 2.

This is an artifact of how your FSM is written. Generally, in order to "command" something to process data from an FSM, you would use `m.d.comb += command.eq(...)`, and in order to capture the outputs and store them in the module whose behavior is defined by the FSM, you would use `m.d.sync += data.eq(...)`. [Playground example](https://amaranth-lang.org/play/#eyJhdiI6InYwLjUuMiIsInMiOiJmcm9tIGFtYXJhbnRoIGltcG9ydCAqXG5mcm9tIGFtYXJhbnRoLnNpbSBpbXBvcnQgU2ltdWxhdG9yXG5mcm9tIGFtYXJhbnRoLmJhY2sgaW1wb3J0IHJ0bGlsLCB2ZXJpbG9nXG5mcm9tIGFtYXJhbnRoLmxpYi5tZW1vcnkgaW1wb3J0IE1lbW9yeVxuZnJvbSBhbWFyYW50aC5saWIgaW1wb3J0IHdpcmluZ1xuZnJvbSBhbWFyYW50aC5saWIud2lyaW5nIGltcG9ydCBJbiwgT3V0XG5pbXBvcnQgYW1hcmFudGhfcGxheWdyb3VuZFxuXG5tID0gTW9kdWxlKClcbm0uc3VibW9kdWxlcy5teW1lbW9yeSA9IG15bWVtb3J5ID0gTWVtb3J5KHNoYXBlPXVuc2lnbmVkKDMyKSwgZGVwdGg9OCwgaW5pdD1baSBmb3IgaSBpbiByYW5nZSg4KV0pXG5cbnJlYWRfcG9ydCA9IG15bWVtb3J5LnJlYWRfcG9ydChkb21haW49XCJzeW5jXCIpXG53cml0ZV9wb3J0ID0gbXltZW1vcnkud3JpdGVfcG9ydChkb21haW49XCJzeW5jXCIpXG5cbnN0YXRlID0gU2lnbmFsKHVuc2lnbmVkKDIpKVxuY29tcHV0YXRpb24gPSBTaWduYWwodW5zaWduZWQoMzIpKVxuXG53aXRoIG0uSWYoc3RhdGUgPT0gMCk6XG4gICAgbS5kLmNvbWIgKz0gcmVhZF9wb3J0LmFkZHIuZXEoNSlcbiAgICBtLmQuY29tYiArPSByZWFkX3BvcnQuZW4uZXEoMSlcbiAgICBtLmQuc3luYyArPSBzdGF0ZS5lcSgxKVxuICAgIG0uZC5zeW5jICs9IFByaW50KFwiU3RhdGUgMCAtIHJlYWQgZGF0YTogXCIsIHJlYWRfcG9ydC5kYXRhKVxud2l0aCBtLkVsaWYoc3RhdGUgPT0gMSk6XG4gICAgbS5kLmNvbWIgKz0gcmVhZF9wb3J0LmVuLmVxKDApXG4gICAgbS5kLnN5bmMgKz0gc3RhdGUuZXEoMilcbiAgICBtLmQuc3luYyArPSBQcmludChcIlN0YXRlIDEgLSByZWFkIGRhdGE6IFwiLCByZWFkX3BvcnQuZGF0YSlcbndpdGggbS5FbGlmKHN0YXRlID09IDIpOlxuICAgICMgZGF0YSBjYW4gb25seSBiZSB1c2VkIG5vdyB0byBjb21wdXRlIHRoaW5ncywgbGlrZSBmb3IgZXhhbXBsZTpcbiAgICBtLmQuc3luYyArPSBjb21wdXRhdGlvbi5lcShyZWFkX3BvcnQuZGF0YSAqIDMgKyAyKVxuICAgICMgb3IgdG8gZG8gYnJhbmNoaW5nIGxpa2UgdGhpczpcbiAgICAjIHdpdGggbS5JZihyZWFkX3BvcnQuZGF0YSA9PSAuLi4pOlxuICAgIG0uZC5zeW5jICs9IFByaW50KFwiU3RhdGUgMiAtIHJlYWQgZGF0YTogXCIsIHJlYWRfcG9ydC5kYXRhKVxuXG5hc3luYyBkZWYgYmVuY2goY3R4KTpcbiAgICBhd2FpdCBjdHgudGljaygpLnJlcGVhdCgzKVxuXG5zaW0gPSBTaW11bGF0b3IobSlcbnNpbS5hZGRfY2xvY2soMWUtNilcbnNpbS5hZGRfdGVzdGJlbmNoKGJlbmNoKVxuI3dpdGggYW1hcmFudGhfcGxheWdyb3VuZC5zaG93X3dhdmVmb3JtcyhzaW0pOlxuc2ltLnJ1bigpXG5cbmFtYXJhbnRoX3BsYXlncm91bmQuc2hvd192ZXJpbG9nKHZlcmlsb2cuY29udmVydChtLCBwb3J0cz1bXSkpXG4ifQ==).

whitequark, 2024-10-02:
> But what I'd really like to do is to already use the read data in state 0. In Verilog, I could do that: [snip]

This would usually create an asynchronous port, and wouldn't synthesize to a BRAM.

epkRichi, 2024-10-02:
Thanks for the updated example. I never even thought of using an FSM with combinational logic like that. Seems like I can keep on using Amaranth then :)

> This would usually create an asynchronous port, and wouldn't synthesize to a BRAM.

So in Verilog I would also need to wait one cycle before using the read data if I want the array to be synthesized as BRAM, meaning that using a memory as a submodule (like Amaranth's `lib.memory.Memory`) actually doesn't slow anything down - right?

whitequark, 2024-10-02:
> meaning that using a memory as a submodule (like Amaranth's `lib.memory.Memory`) actually doesn't slow anything down - right?

Yes.

epkRichi, 2024-10-02:
Alright, thanks a lot for your help, you're awesome!

closed 2024-10-02 by epkRichi

whitequark, 2024-10-02:
Happy to help!