amaranth-lang/amaranth #1534: Formatting of enums without `shape` in a `data.StructLayout` does not work

rroohhh, 2024-10-14:
Take the following example:
```python
#!/usr/bin/env python3

from amaranth import *
from amaranth.lib.enum import Enum
from amaranth.lib import data
from amaranth.back.rtlil import convert

m = Module()

class test(Enum):
    a = 0
    b = 1

class B(data.Struct):
    a: test

a = Signal(test)
b = Signal(B)
input1 = Signal()
input2 = Signal()
m.d.comb += a.eq(input1)
m.d.comb += b.eq(input2)

print(convert(m, ports=(input1, input2,)))
```
This outputs (simplified)
```
module \top
  attribute \enum_base_type "test"
  attribute \enum_value_0 "a"
  attribute \enum_value_1 "b"
  wire width 1 \a

  wire width 1 \b

  attribute \enum_base_type "test"
  attribute \enum_value_0 "a"
  attribute \enum_value_1 "b"
  wire width 1 input 0  \input1

  wire width 1 input 1  \input2

  wire width 1 \b.a

connect \a \input1 [0]
connect \b \input2 [0]
connect \b.a \input2 [0]

end
```
I would have expected the `\b.a` wire to also have the `enum_{base_type,value_*}` attributes, but these are only generated when I give my enum an explicit `shape=...`.

rroohhh, 2024-10-14:
This can of course be fixed by making the `data.Struct`, etc parts add special casing for "plain" `Enum`s, however I am not sure if this is the most elegant way.

rroohhh, 2024-10-14:
Do you want a PR implementing my suggested solution?

whitequark, 2024-10-14:
Let's ask @wanda-phi, who implemented this originally.