amaranth-lang/amaranth #1539: Iterating over `View`s of structures

gatecat, 2024-10-28:
Currently, attempting to iterate over the fields of a structure like:

```python
from amaranth import *
from amaranth.lib import stream, wiring, data
from amaranth.lib.wiring import In, Out

class StructTest(wiring.Component):

    data: In(data.StructLayout({
        "a": 48,
        "b": 48,
    }))

    def elaborate(self, platform):
        m = Module()

        for name, field in self.data:
            m.d.sync += Print(f"field {name} ", field)
        return m

if __name__ == '__main__':
    from amaranth.cli import main
    main(StructTest())
```

Fails with a somewhat hard to interpret exception:

```
  File "/home/gatecat/scratching-post/cxxrtl-tap/.venv/lib/python3.12/site-packages/amaranth/lib/data.py", line 833, in __getitem__
    field = self.__layout[key]
            ~~~~~~~~~~~~~^^^^^
  File "/home/gatecat/scratching-post/cxxrtl-tap/.venv/lib/python3.12/site-packages/amaranth/lib/data.py", line 344, in __getitem__
    return self._fields[key]
           ~~~~~~~~~~~~^^^^^
KeyError: 0
```

Either this should fail with a clearer error that it's not a supported pattern, or be supported.

whitequark, 2024-10-30:
Yeah, we could definitely improve it.

whitequark, 2024-10-30:
I think what you're seeing is Python's default iteration behavior for when you don't have an explicit `__iter__` but have a `__getitem__`; in this case it assumes that the collection is an array.

@wanda-phi Should we just implement `__iter__` that errors out?

@gatecat You're probably looking for this ([playground](https://amaranth-lang.org/play/#eyJhdiI6InYwLjUuMiIsInMiOiJmcm9tIGFtYXJhbnRoIGltcG9ydCAqXG5mcm9tIGFtYXJhbnRoLmxpYiBpbXBvcnQgc3RyZWFtLCB3aXJpbmcsIGRhdGFcbmZyb20gYW1hcmFudGgubGliLndpcmluZyBpbXBvcnQgSW4sIE91dFxuXG5cbmNsYXNzIFN0cnVjdFRlc3Qod2lyaW5nLkNvbXBvbmVudCk6XG5cbiAgICBkYXRhOiBJbihkYXRhLlN0cnVjdExheW91dCh7XG4gICAgICAgIFwiYVwiOiA0OCxcbiAgICAgICAgXCJiXCI6IDQ4LFxuICAgIH0pKVxuXG4gICAgZGVmIGVsYWJvcmF0ZShzZWxmLCBwbGF0Zm9ybSk6XG4gICAgICAgIG0gPSBNb2R1bGUoKVxuXG4gICAgICAgIGZvciBuYW1lLCBmaWVsZCBpbiBzZWxmLmRhdGEuc2hhcGUoKTpcbiAgICAgICAgICAgIG0uZC5zeW5jICs9IFByaW50KGZcImZpZWxkIHtuYW1lfSBcIiwgZmllbGQpXG4gICAgICAgIHJldHVybiBtXG5cblxuZnJvbSBhbWFyYW50aC5zaW0gaW1wb3J0IFNpbXVsYXRvclxuc2ltID0gU2ltdWxhdG9yKFN0cnVjdFRlc3QoKSlcbnNpbS5hZGRfY2xvY2soMWUtNilcbnNpbS5ydW5fdW50aWwoMmUtNikifQ==)):

```amaranth
for name, field in self.data.shape():
    m.d.sync += Print(f"field {name} ", field)
```

wanda-phi, 2024-10-30:
> I think what you're seeing is Python's default iteration behavior for when you don't have an explicit `__iter__` but have a `__getitem__`; in this case it assumes that the collection is an array.
> 
> @wanda-phi Should we just implement `__iter__` that errors out?

I tried directly implementing `__iter__`; the problem with it is that the `View` then becomes `instanceof` `Iterable`, and the check in `Cat` tries to iterate it instead of treating it as a `Value`, causing sadness

I think we have discussed this at some point, but I don't really remember the details?

whitequark, 2024-10-30:
Oh, ugh. I vaguely remember this as well. And I don't think we can reasonably throw something other than `KeyError` here, can we?

wanda-phi, 2024-10-30:
since this is the `__getitem__` implementation being called, no, not really

whitequark, 2024-10-30:
@gatecat Sad as it is I think our hands are tied by Python here. :/

closed 2024-10-30 by whitequark