amaranth-lang/amaranth #1558: Confusing signal names in VCD / GTKWave depending on the order signals are referenced in elaboration code

ajburgess, 2025-02-01:
When I run a simulation for a component and view the resulting VCD file in GTKWave, and that component makes use of other components (sub-modules) that share some of the same signal names, then the names appearing in GTGWave can be confusing, and make it hard to pick the relevant signals I'm interested in. For signals with the same name, one signal will have the original name, and the others will have '$n' appended. Originally I assumed that the main component's signal names would be the ones without '$n' appended, but that's not always the case. Until I realised this, I spent a lot of time trying to debug a deisgn based on the simulation wave traces when in fact there was nothing wrong with it! (As a beginner, I find viewing traces is the easiest way to check I've understood Amaranth concepts correctly) - so I suspect other people in a simular situation to me could make the same mistake / be confused in the same way.

Here's the simplest design I could think of that demonstrates the issue, so please ignore the fact that it's a kind of pointless circuit!

```python
# Using Amaranth version 0.5.4

from amaranth import *
from amaranth.lib import wiring
from amaranth.lib.wiring import In, Out
from amaranth.sim import Simulator, Period

class ComponentB(wiring.Component):
    i_data: In(1)
    o_data: Out(1)

    def elaborate(self, platform):
        m = Module()
        m.d.sync += self.o_data.eq(self.i_data)
        return m

class ComponentA(wiring.Component):
    i_data: In(1)
    o_data: Out(1)

    def elaborate(self, platform):
        m = Module()
        m.submodules.b = b = ComponentB()
        m.d.sync += b.i_data.eq(self.i_data)
        m.d.sync += self.o_data.eq(b.o_data)
        return m
    
dut = ComponentA()

async def testbench(ctx):
    ctx.set(dut.i_data, 0)
    await ctx.tick().repeat(5)
    ctx.set(dut.i_data, 1)
    await ctx.tick().repeat(5)
    ctx.set(dut.i_data, 0)
    await ctx.tick().repeat(5)

sim = Simulator(dut)
sim.add_clock(Period(MHz=25))
sim.add_testbench(testbench)
with sim.write_vcd("component.vcd"):
    sim.run()
```

Which generates the following wave traces in GTKWave.

[image]

As can be seen, `i_data$3` is actually the signal in the top component, and `i_data` is actaully the signal in the sub-component. But signal `o_data` is from the main component and `o_data$5` is from the sub-component - so not consistent. I eventually figured it's to do with the order the signals are referenced in the elaboration of the main component. And because when wiring the input from one componen to the input of a sub-component, you mention the sub-component first, then it's the sub-component's signal that takes the "original" name, and the main component's signal that gets the "duplicate" name with the '$n' appended.

So my question is, am I doing something wrong or missing something simple that would enable the main component's signal names to always be the "original" ones rather than the "$n" ones (which would make it easier to pick the traces I'm interested in)? Or is there any way I can only see the main component's signals under "top" in the traces, and if I'm interested in a sub-modules signals then I can drill down into that sub-module in the "tree" in GTKWave and pick its signals that way?

[I've tried various approaches using wiring.Component vs Elaboratable, using different wave viewers, etc, but always get the same result. I also appreciate I could just try to redesign all the signal names used in various component so there is never a duplicate, but that seems like I'm producing a sub-optinal design just to make the simulation easier.]

ajburgess, 2025-02-01:
I just updated the code sample above as I'd  been experimenting with version 0.5.4 verses the latest build version and had pasted the wrong one. The results are the same either way.

whitequark, 2025-02-01:
This is a known issue. I don't recall what exactly is the current status of it at the moment.

ajburgess, 2025-02-01:
Thank you for the quick response. That's good to know it's not something I can change through my code right now. I've have another look again through previous issues / discussions and if I can find it this time I'll add a link to it. I'll close the issue though.

closed 2025-02-01 by ajburgess