amaranth-lang/amaranth #1569: Missing return in elaborate results in cryptic error

robtaylor, 2025-03-04:
A missing return gives a useful warning: 

`UserWarning: .elaborate() returned None; missing return statement?`

but this is then followed by a cryptic error:
 
`TypeError: Object None is not an 'Elaboratable' nor 'Fragment'`

Would it be better to raise an error instead of a warning?

whitequark, 2025-03-04:
That seems reasonable. I'll review the code to understand why it is using a warning and then if there is no good reason I'll upgrade it to an error.

whitequark, 2025-03-05:
So there actually *is* a reason for using a warning here. This is because it is possible in Python to override the file and line attached to a warning:

https://github.com/amaranth-lang/amaranth/blob/cf4a29d3a29fd316207aa77c5b63f991a4e81756/amaranth/hdl/_ir.py#L69-L73

But it's not possible to add them to the backtrace without some CPython-specific hacks.

whitequark, 2025-03-05:
I've changed it to:

```
TypeError: Object None is not an 'Elaboratable' nor 'Fragment', returned by /home/whitequark/Projects/amaranth/tests/test_hdl_ir.py:19
```

closed 2025-03-05 by whitequark

robtaylor, 2025-03-05:
Nice! Its a shame to loose the message about it possibly being a missed return though, as thats the likely beginner issue.

commit 199c34ff (2025-03-05): hdl.ir: improve error raised when returning None from `elaborate`.

Fixes #1569.
PR #1570 (2025-03-05): Improve error raised when returning None from `elaborate`
Fixes #1569.